OpenTabletDriver/OpenTabletDriver #1448: Gaomon S56K ReportParser problem

Patriot720, 2021-09-19:
## Description
idk if it's my specific gaomon s56k problem or not but
https://github.com/OpenTabletDriver/OpenTabletDriver/blob/04adf913cdaa282ce97d86f0e8ebee12264ef17e/OpenTabletDriver/Vendors/UCLogic/UCLogicReportParser.cs#L6-L16

[tablet-data.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7191270/tablet-data.txt)
That bit is only 00 upon entering tablet "interactive area"
And most of the time it's not
``` json
{
      "VendorID": 9580,
      "ProductID": 110,
      "InputReportLength": 8,
      "OutputReportLength": null,
      "ReportParser": "OpenTabletDriver.Tablet.TabletReportParser",
      "FeatureInitReport": null,
      "OutputInitReport": null,
      "DeviceStrings": {
        "201": "HUION_T156_\\d{6}$"
      },
      "InitializationStrings": [
        200
      ]
}
```
Works fine if i set TabletReportParser manually in the s56k config
[Vid](https://puu.sh/IcqOF/9b0a219147.mp4)

## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System         Arch linux|
| OpenTabletDriver Version OpenTabletDriver v0.5.3.1-r460.g04adf913 |
| Tablet                   Gaomon S56K |

X9VoiD, 2021-09-19:
There's no aux data recorded in the tablet-data.

Patriot720, 2021-09-19:
> There's no aux data recorded in the tablet-data.

There are no buttons on that tablet why would there ever be aux data, unless i misunderstand what aux here means.

Also that's what Tablet debugger data recording outputs into a file, at least on my system.
Only raw tablet data there 
[image]

PR #1681 (2021-12-06): Gaomon S56K: Use TabletReportParser
Gaomon S56K has no aux keys, therefore it makes no sense to use a parser that checks for aux keys and then falling back on TabletReport.

Closes #1448
commit b74dbce3 (2021-12-06): Gaomon S56K: Use TabletReportParser

Gaomon S56K has no aux keys, therefore it makes no sense to use a parser
that checks for aux keys and then falling back on TabletReport.

Closes #1448
PR #1679 (2021-12-13): Update configurations and parsers
- Closes #1101
- Closes #1282
- Closes #1338
- Closes #1401
- Closes #1435
- Closes #1448
- Closes #1500
- Closes #1545
closed 2021-12-13 by InfinityGhost