OpenTabletDriver/OpenTabletDriver #1449: OpenTabletDriver not searching for UC-Logic configurations when detecting a tablet

mxmint, 2021-09-19:
## Description

App reports no detected tablet either on startup or upon selecting the "Detect tablet" option. The console printout shows search for Wacom, VEIKK, Huion, XP-Pen and Gaomon model configs, then aborts with "No tablet found". In `lsusb` the tablet I'm using is listed as "UC-Logic Technology Corp. Tablet PF1209", which is on the list of supported devices.
## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System | Linux (EndeavourOS)
| OpenTabletDriver version | 0.5.3.3
| Tablet | UC-Logic PF1209 (Monoprice MP1290-A1)

SnowLire, 2021-09-19:
Your tablet is supported only on the -git version of OpenTabletDriver for now. You could download that, however there are some bugs, you can't download any plugins from the plugin manager, and most plugins don't work (#1407)

mxmint, 2021-09-19:
Unfortunately the git version crashes when detecting a tablet, with this printout:
```
Exception in Gtk# callback delegate
  Note: Applications can use GLib.ExceptionManager.UnhandledException to handle the exception.
StreamJsonRpc.RemoteMethodNotFoundException: No method by the name 'GetTablets' is found.
   at StreamJsonRpc.JsonRpc.InvokeCoreAsync[TResult](RequestId id, String targetName, IReadOnlyList`1 arguments, IReadOnlyList`1 positionalArgumentDeclaredTypes, IReadOnlyDictionary`2 namedArgumentDeclaredTypes, CancellationToken cancellationToken, Boolean isParameterObject)
   at OpenTabletDriver.UX.Controls.Output.OutputModeEditor.<>c__DisplayClass1_0.<<UpdateTablet>b__0>d.MoveNext() in /home/user/.cache/yay/opentabletdriver-git/src/OpenTabletDriver/OpenTabletDriver.UX/Controls/Output/OutputModeEditor.cs:line 49
--- End of stack trace from previous location ---
   at System.Threading.Tasks.Task.<>c.<ThrowAsync>b__140_0(Object state)
   at Eto.GtkSharp.Forms.GtkSynchronizationContext.<>c__DisplayClass0_0.<Post>b__0()
   at GLib.Idle.IdleProxy.Handler()
   at GLib.ExceptionManager.RaiseUnhandledException(Exception e, Boolean is_terminal)
   at GLib.Idle.IdleProxy.Handler()
   at Eto.GtkSharp.Forms.ApplicationHandler.Run()
   at Eto.Forms.Application.Run(Form mainForm)
   at OpenTabletDriver.UX.App.Run(String platform, String[] args) in /home/user/.cache/yay/opentabletdriver-git/src/OpenTabletDriver/OpenTabletDriver.UX/App.cs:line 57
   at OpenTabletDriver.UX.Gtk.Program.Main(String[] args) in /home/user/.cache/yay/opentabletdriver-git/src/OpenTabletDriver/OpenTabletDriver.UX.Gtk/Program.cs:line 10

```

X9VoiD, 2021-09-19:
You're trying to run UX with an outdated daemon running in the background. Make sure you're running the latest version of the daemon as well.

mxmint, 2021-09-19:
Could you elaborate on how to do that, please? I've simply installed otd-git from the AUR, I don't really know how to upgrade its individual components.

jamesbt365, 2021-09-19:
you need to run systemctl --user restart opentabletdriver and then restart the ux

gonX, 2021-12-06:
Closing issue as abandoned. I assume it works now, given that `opentabletdriver-git` has been reported working for many people now.

closed 2021-12-06 by gonX