OpenTabletDriver/OpenTabletDriver #1473: Eraser end is barely usable on IntuosV1 parser (IToolReport's mostly unused)

gonX, 2021-10-01:
## Description

#1255 implements `IToolReport`, but does not make OTD aware when the eraser end is present on `IntuosV1` parser tablets - leading to eraser-end bindings not working in OTD for these tablets.

These tablets give a tool report as the very first report on proximity and another one on out of range. Example of in-range:
[image]
(provoked by moving tool in slowly from the side)

Linux' Artist mode is not hugely affected as it caches values from `IEraserReport`s (which `IntuosV1ToolReport` implements because there currently isn't any good tool handling) and includes it on subsequent position updates (#1232), though the eraser end will incorrectly (but often desirably) use the tip binding instead.

One fix could be to annotate the tool report onto the non-tool reports. It can easily be done in the parser, but I have feeling it isn't the correct approach. 

The tested tablet is a GD-0608, but it is expected that the other tablets using this parser acts the same way.
Fixing this would also fix the eraser issues detailed in #1428 

## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System         | All
| OpenTabletDriver Version | Latest (4c4ed2891e63ad829322cb7ff871a520864360b2)
| Tablet                   | Wacom GD-0608 and other `IntuosV1`-parser tablets

gonX, 2021-11-30:
Setting as low priority as Linux Artist Mode has a workaround and I imagine not a lot of people are using IntuosV1 tablets on Windows.

X9VoiD, 2022-12-31:
Fixed by #2557

closed 2022-12-31 by X9VoiD