OpenTabletDriver/OpenTabletDriver #1476: Driver doesn't work on newest version of arch kernel.

Shchepan, 2021-10-01:
## Description

The driver stopped working after updating linux-rt kernel from version 5.10.59.52 to 5.10.65.53. 
The same problem appears with newest regular Arch kernel. 
After changing rt-lts kernel back to version 5.10.59.52  everything seems to work flawlessly. 
Running otd from terminal outputs
`OpenTabletDriver Daemon is already running.`
but launching otd-gui gives
`(OpenTabletDriver.UX.Gtk:1193): libnotify-WARNING **: 13:04:50.555: Failed to connect to proxy`
and it can't detect my tablet.
The driver was installed through AUR, both regular and -git version have the same problem.
I apologize for my incompetence in writing this kind of thing. It's my first time writing bug report on github.
## System Information:

| Name             | Value |
| Operating System |   Arch Linux x86_64     |
| OpenTabletDriver Version |
| Tablet   |   Gaomon S620             |

jamesbt365, 2021-10-01:
restart the daemon with systemctl?

Shchepan, 2021-10-01:
> restart the daemon with systemctl?
doing `systemctl daemon-reload`
doesn't seem to help, if that's what you meant.

coinworld, 2021-10-01:
Running `otd` without arguments will just run OTD daemon.
To check if OTD daemon is running, you need to use `systemctl --user status opentabletdriver`
and you can restart daemon with `systemctl --user restart opentabletdriver` if you want to try what james suggested.

Upload diagnotics from `Help -> Export diagnotics` here if you couldn't fix with restarting.

Shchepan, 2021-10-01:
the  Help -> Export diagnotics
is attached

systemctl --user status opentabletdriver output
`* opentabletdriver.service - OpenTabletDriver Daemon
     Loaded: loaded (/usr/lib/systemd/user/opentabletdriver.service; enabled; vendor preset: enabled)
     Active: active (running) since Fri 2021-10-01 13:36:24 CEST; 13min ago
   Main PID: 860 (otd)
[diagno.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7266897/diagno.txt)
      Tasks: 10 (limit: 9458)
     Memory: 66.7M
        CPU: 1.469s
     CGroup: /user.slice/user-1000.slice/user@1000.service/app.slice/opentabletdriver.service
             |-860 /bin/bash /usr/bin/otd
             `-861 dotnet ./OpenTabletDriver.Daemon.dll -c ./Configurations

Oct 01 13:36:25 heh otd[861]: [Settings:Info]        Tablet area: [0x0@<0, 0>:0<C2><B0>],
Oct 01 13:36:25 heh otd[861]: [Settings:Info]        Clipping: Disabled
Oct 01 13:36:25 heh otd[861]: [Settings:Info]        Ignoring reports outside area: Disabled
Oct 01 13:36:25 heh otd[861]: [Settings:Info]        Tip Binding: []@0%
Oct 01 13:36:25 heh otd[861]: [Settings:Info]        Pen Bindings: [0, ], [1, ]
Oct 01 13:36:25 heh otd[861]: [Settings:Info]        Express Key Bindings: [0, ], [1, ], [2, ], [3, ], [4, ], [5, ], [6, ], [7, ]
Oct 01 13:36:55 heh otd[861]: [IPC:Debug]        Connected to a client.
Oct 01 13:37:00 heh otd[861]: [IPC:Debug]        Disconnected from a client.
Oct 01 13:45:17 heh otd[861]: [IPC:Debug]        Connected to a client.
Oct 01 13:46:00 heh otd[861]: [IPC:Debug]        Disconnected from a client.`

Shchepan, 2021-10-01:
[diagno.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7266905/diagno.txt)

coinworld, 2021-10-01:
Can you try exporting diagnotics again on the working kernel?

Shchepan, 2021-10-01:
[diagno2.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7267111/diagno2.txt)

coinworld, 2021-10-01:
It seems something went wrong in your environment. OTD has been just fine on my pc with 5.14.x

Shchepan, 2021-10-01:
Ok, at least could you tell me what could have made it that way, of if you have any speculations? I'm not very familiar with how daemons and stuff like that works.

jamesbt365, 2021-10-01:
Well none of your devices pop up, not even a mouse or keyboard

Shchepan, 2021-10-02:
I'm going to close this since it's seems like my issue rather than bug in program itself. Idk what I'm going to do now, I probably should switch to something with no systemd. Who knows. Anyway thanks for help.

closed 2021-10-02 by Shchepan