OpenTabletDriver/OpenTabletDriver #1477: Add driver for trust tablets

TheRealOne78, 2021-10-03:
## Description

Driver for Trust 21794

OpenTabletDriver seems to not have support for any trust tablets. There are quite some trust tablets users in the world, yet no drivers for linux.

X9VoiD, 2021-10-03:
Connect your tablet, OTD -> Export Diagnostics.

jamesbt365, 2021-10-03:
if you own one of these you would need to start by giving us information of one, we need a daig from within otd, you can get one by going to help > export diagnosis and send it here.

TheRealOne78, 2021-10-03:
wow.. that was definetly fast reply...

Back to main idea, here are the diagnostics:
[diagnostics.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7273191/diagnostics.txt)

```
Driver specs: 
Height of main product (in mm)
238 mm
Width of main product (in mm)
358 mm
Depth of main product (in mm)
```
Official tablet specs: https://www.trust.com/en/product/21794-panora-widescreen-graphic-tablet

Has a special driver for windows that isn't very necessary.

Linux seems to have difficulties with the tablet.

X9VoiD, 2021-10-03:
Download and extract this: [generic_prep_scripts.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7273215/generic_prep_scripts.zip)

Then run `generic_prep_system.sh` as root. This allows OTD to read from all devices, you may always run `generic_unprep_system.sh` to undo the changes done by the first script.

Redo the diagnostics after running the script.

TheRealOne78, 2021-10-03:
[diagv2.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7273222/diagv2.txt)

X9VoiD, 2021-10-03:
Go to OTD -> Tablet -> Device string reader

VID: 5215
PID: 530

Then press dump all.

TheRealOne78, 2021-10-03:
[dump.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7273272/dump.txt)

X9VoiD, 2021-10-03:
Extract [Panora.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7273284/Panora.zip) into  `<otd_data_folder>/Configurations/Trust/Panora.json`

If you installed OTD through a package manager then `<otd_data_folder>` will be in `/usr/share/opentabletdriver`

Go to OTD -> Tablet -> Detect tablet. It may or may not work, the configuration is not finalized yet.

TheRealOne78, 2021-10-03:
There is a new title in the title bar, ``OTD - Trust Panora``, it doesn't recognise the tablet tho...
Visualizer gets output.
[image]
[image]

X9VoiD, 2021-10-03:
File -> Reset to default

TheRealOne78, 2021-10-03:
Reseted it, title changed to no tablet detected, tablet debuger has no readings.
[image]

X9VoiD, 2021-10-03:
Export diagnostics

TheRealOne78, 2021-10-03:
[diag.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7273459/diag.txt)

jamesbt365, 2021-10-03:
https://github.com/OpenTabletDriver/OpenTabletDriver/wiki/Linux-FAQ#the-driver-fails-to-open-the-tablet-deviceioexception

do this

TheRealOne78, 2021-10-03:
[cli.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7273465/cli.txt)

X9VoiD, 2021-10-03:
No, that won't fix the problem. Did you run the `unprep` script?

TheRealOne78, 2021-10-03:
yes

TheRealOne78, 2021-10-03:
so... rerun prep then reruning unprep, or just run prep.sh?

X9VoiD, 2021-10-03:
Do not run `unprep` until full support is merged in OTD.

TheRealOne78, 2021-10-03:
ok...
```
❯ sudo bash generic_prep_system.sh
[sudo] password for therealone: 
Installed /etc/udev/rules.d/95-otd_devices.rules
```

TheRealOne78, 2021-10-03:
I see there is no reply... so I am wondering if you expect me to try again everything from ``gen_prep_scripts`` until the last step since that steps were done without ``prep.sh`` and now I rerun-ed prep.sh.

jamesbt365, 2021-10-03:
use prep.sh and do everything up to now, dont unprep it until we have completely done with getting it working with otd.

TheRealOne78, 2021-10-03:
understood

TheRealOne78, 2021-10-03:
```
Go to OTD -> Tablet -> Device string reader

VID: 5215
PID: 530

Then press "Dump all".
```
[dump2.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7273809/dump2.txt)

TheRealOne78, 2021-10-03:
```
Extract Panora.zip into <otd_data_folder>/Configurations/Trust/Panora.json

If you installed OTD through a package manager then <otd_data_folder> will be in /usr/share/opentabletdriver

Go to OTD -> Tablet -> Detect tablet. It may or may not work, the configuration is not finalized yet.
```
It seems like this might need to depend in dump2.txt, but will try this anyway.

TheRealOne78, 2021-10-03:
```
File -> Reset to default
```
done.
[image]

```
Export diagnostics
```

[diag-nounprep.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7273970/diag-nounprep.txt)

TheRealOne78, 2021-10-03:
```
https://github.com/OpenTabletDriver/OpenTabletDriver/wiki/Linux-FAQ#the-driver-fails-to-open-the-tablet-deviceioexception

do this
```
```
❯ git clone https://github.com/InfinityGhost/OpenTabletDriver-udev.git
cd OpenTabletDriver-udev
git submodule update --init --remote
./build.sh
sudo mv ./build/99-opentabletdriver.rules /etc/udev/rules.d/99-opentabletdriver.rules
sudo udevadm control --reload-rules
cd ..
rm -rf OpenTabletDriver
rm -rf OpenTabletDriver-udev

Cloning into 'OpenTabletDriver-udev'...
remote: Enumerating objects: 175, done.
remote: Counting objects: 100% (175/175), done.
remote: Compressing objects: 100% (108/108), done.
remote: Total 175 (delta 88), reused 131 (delta 54), pack-reused 0
Receiving objects: 100% (175/175), 40.58 KiB | 1.50 MiB/s, done.
Resolving deltas: 100% (88/88), done.
Submodule '.modules/OpenTabletDriver' (https://github.com/InfinityGhost/OpenTabletDriver.git) registered for path '.modules/OpenTabletDriver'
Cloning into '/home/therealone/OpenTabletDriver-udev/.modules/OpenTabletDriver'...
Submodule path '.modules/OpenTabletDriver': checked out '4c4ed2891e63ad829322cb7ff871a520864360b2'
Writing all rules to '/build/99-opentabletdriver.rules'...
KERNEL=="uinput", SUBSYSTEM=="misc", TAG+="uaccess", OPTIONS+="static_node=uinput"
# Artisul M0610 Pro
SUBSYSTEM=="hidraw", ATTRS{idVendor}=="5543", ATTRS{idProduct}=="0081", MODE="0666"
SUBSYSTEM=="usb", ATTRS{idVendor}=="5543", ATTRS{idProduct}=="0081", MODE="0666"
SUBSYSTEM=="input", ATTRS{idVendor}=="5543", ATTRS{idProduct}=="0081", ENV{LIBINPUT_IGNORE_DEVICE}="1"
# UC-Logic PF1209
SUBSYSTEM=="hidraw", ATTRS{idVendor}=="5543", ATTRS{idProduct}=="0042", MODE="0666"
SUBSYSTEM=="usb", ATTRS{idVendor}=="5543", ATTRS{idProduct}=="0042", MODE="0666"
SUBSYSTEM=="input", ATTRS{idVendor}=="5543", ATTRS{idProduct}=="0042", ENV{LIBINPUT_IGNORE_DEVICE}="1"
# UC-Logic 1060N
SUBSYSTEM=="hidraw", ATTRS{idVendor}=="5543", ATTRS{idProduct}=="0081", MODE="0666"
SUBSYSTEM=="usb", ATTRS{idVendor}=="5543", ATTRS{idProduct}=="0081", MODE="0666"
SUBSYSTEM=="input", ATTRS{idVendor}=="5543", ATTRS{idProduct}=="0081", ENV{LIBINPUT_IGNORE_DEVICE}="1"
# XP-Pen Artist 12 Pro
SUBSYSTEM=="hidraw", ATTRS{idVendor}=="28bd", ATTRS{idProduct}=="091f", MODE="0666"
SUBSYSTEM=="usb", ATTRS{idVendor}=="28bd", ATTRS{idProduct}=="091f", MODE="0666"
SUBSYSTEM=="input", ATTRS{idVendor}=="28bd", ATTRS{idProduct}=="091f", ENV{LIBINPUT_IGNORE_DEVICE}="1"
# XP-Pen Star G640 V2
SUBSYSTEM=="hidraw", ATTRS{idVendor}=="28bd", ATTRS{idProduct}=="0914", MODE="0666"
SUBSYSTEM=="usb", ATTRS{idVendor}=="28bd", ATTRS{idProduct}=="0914", MODE="0666"
SUBSYSTEM=="input", ATTRS{idVendor}=="28bd", ATTRS{idProduct}=="0914", ENV{LIBINPUT_IGNORE_DEVICE}="1"
# XP-Pen Deco Mini7
SUBSYSTEM=="hidraw", ATTRS{idVendor}=="28bd", ATTRS{idProduct}=="0928", MODE="0666"
SUBSYSTEM=="usb", ATTRS{idVendor}=="28bd", ATTRS{idProduct}=="0928", MODE="0666"
SUBSYSTEM=="input", ATTRS{idVendor}=="28bd", ATTRS{idProduct}=="0928", ENV{LIBINPUT_IGNORE_DEVICE}="1"
# XP-Pen Star G430S V2
SUBSYSTEM=="hidraw", ATTRS{idVendor}=="28bd", ATTRS{idProduct}=="0913", MODE="0666"
SUBSYSTEM=="usb", ATTRS{idVendor}=="28bd", ATTRS{idProduct}=="0913", MODE="0666"
SUBSYSTEM=="input", ATTRS{idVendor}=="28bd", ATTRS{idProduct}=="0913", ENV{LIBINPUT_IGNORE_DEVICE}="1"
# XP-Pen Deco Pro Medium
SUBSYSTEM=="hidraw", ATTRS{idVendor}=="28bd", ATTRS{idProduct}=="0904", MODE="0666"
SUBSYSTEM=="usb", ATTRS{idVendor}=="28bd", ATTRS{idProduct}=="0904", MODE="0666"
SUBSYSTEM=="input", ATTRS{idVendor}=="28bd", ATTRS{idProduct}=="0904", ENV{LIBINPUT_IGNORE_DEVICE}="1"
# XP-Pen Artist 15.6 Pro
SUBSYSTEM=="hidraw", ATTRS{idVendor}=="28bd", ATTRS{idProduct}=="090d", MODE="0666"
SUBSYSTEM=="usb", ATTRS{idVendor}=="28bd", ATTRS{idProduct}=="090d", MODE="0666"
SUBSYSTEM=="input", ATTRS{idVendor}=="28bd", ATTRS{idProduct}=="090d", ENV{LIBINPUT_IGNORE_DEVICE}="1"
# XP-Pen Star G960S Plus
SUBSYSTEM=="hidraw", ATTRS{idVendor}=="28bd", ATTRS{idProduct}=="0918", MODE="0666"
SUBSYSTEM=="usb", ATTRS{idVendor}=="28bd", ATTRS{idProduct}=="0918", MODE="0666"
SUBSYSTEM=="input", ATTRS{idVendor}=="28bd", ATTRS{idProduct}=="0918", ENV{LIBINPUT_IGNORE_DEVICE}="1"
# XP-Pen Star G640S
SUBSYSTEM=="hidraw", ATTRS{idVendor}=="28bd", ATTRS{idProduct}=="0906", MODE="0666"
[... 463 lines cut ...]
SUBSYSTEM=="hidraw", ATTRS{idVendor}=="056a", ATTRS{idProduct}=="00b5", MODE="0666"
SUBSYSTEM=="usb", ATTRS{idVendor}=="056a", ATTRS{idProduct}=="00b5", MODE="0666"
# Parblo Intangbo S
SUBSYSTEM=="hidraw", ATTRS{idVendor}=="0483", ATTRS{idProduct}=="a014", MODE="0666"
SUBSYSTEM=="usb", ATTRS{idVendor}=="0483", ATTRS{idProduct}=="a014", MODE="0666"
SUBSYSTEM=="input", ATTRS{idVendor}=="0483", ATTRS{idProduct}=="a014", ENV{LIBINPUT_IGNORE_DEVICE}="1"
# Parblo Ninos S
SUBSYSTEM=="hidraw", ATTRS{idVendor}=="0483", ATTRS{idProduct}=="a006", MODE="0666"
SUBSYSTEM=="usb", ATTRS{idVendor}=="0483", ATTRS{idProduct}=="a006", MODE="0666"
SUBSYSTEM=="input", ATTRS{idVendor}=="0483", ATTRS{idProduct}=="a006", ENV{LIBINPUT_IGNORE_DEVICE}="1"
# Parblo Ninos M
SUBSYSTEM=="hidraw", ATTRS{idVendor}=="0483", ATTRS{idProduct}=="a005", MODE="0666"
SUBSYSTEM=="usb", ATTRS{idVendor}=="0483", ATTRS{idProduct}=="a005", MODE="0666"
SUBSYSTEM=="input", ATTRS{idVendor}=="0483", ATTRS{idProduct}=="a005", ENV{LIBINPUT_IGNORE_DEVICE}="1"
# Parblo Intangbo M
SUBSYSTEM=="hidraw", ATTRS{idVendor}=="0483", ATTRS{idProduct}=="a013", MODE="0666"
SUBSYSTEM=="usb", ATTRS{idVendor}=="0483", ATTRS{idProduct}=="a013", MODE="0666"
SUBSYSTEM=="input", ATTRS{idVendor}=="0483", ATTRS{idProduct}=="a013", ENV{LIBINPUT_IGNORE_DEVICE}="1"
Finished writing all rules.
[sudo] password for therealone: 

    ~ ▓▒░·································░▒▓ took 16s   at 06:54:04   ─╮
❯     
```
done.

TheRealOne78, 2022-10-26:
Hi, this issue stalled.

jamesbt365, 2022-10-26:
Update to 0.6.0.4 and i'll port the config and we'll continue from there.

TheRealOne78, 2022-10-27:
Meanwhile I've moved from Ubuntu to Artix with KDE Plasma X11.
Since there is no OpenRC service, I am running `otd` daemon as a user.

I've preped my system and put Panora.json in `/usr/share/OpenTabletDriver/Configurations/Trust/`, but I have to mention I had to create the `Configurations` dir.

For some reason the daemon doesn't detect the tablet and throws this:
```cs
[Detect:Debug]	Searching for tablet 'Trust Panora'
[ArgumentOutOfRangeException:Error]	Value range is [0, 15]. (Parameter 'value')
   at HidSharp.Reports.Units.Unit.DecodeExponent(UInt32 value)
   at HidSharp.Reports.ReportDescriptor.ParseMainData(MainItemTag tag, UInt32 value)
   at HidSharp.Reports.ReportDescriptor.ParseEncodedItem(EncodedItem item)
   at HidSharp.Reports.ReportDescriptor.ParseEncodedItems(IEnumerable`1 items)
   at HidSharp.Reports.ReportDescriptor.ParseRawReportDescriptor(Byte[] buffer)
   at HidSharp.Reports.ReportDescriptor..ctor(Byte[] buffer)
   at HidSharp.Platform.Linux.LinuxHidDevice.TryParseReportDescriptor(ReportDescriptor& parser, Byte[]& reportDescriptor)
   at HidSharp.Platform.Linux.LinuxHidDevice.RequiresGetInfo()
   at HidSharp.Platform.Linux.LinuxHidDevice.GetMaxInputReportLength()
   at OpenTabletDriver.Devices.HidSharpBackend.HidSharpEndpoint.<>c.<get_InputReportLength>b__7_0(HidDevice d)
   at OpenTabletDriver.Extensions.TryGet[TSource,TValue](TSource source, Func`2 predicate, TValue& value)
[ArgumentOutOfRangeException:Error]	Value range is [0, 15]. (Parameter 'value')
   at HidSharp.Reports.Units.Unit.DecodeExponent(UInt32 value)
   at HidSharp.Reports.ReportDescriptor.ParseMainData(MainItemTag tag, UInt32 value)
   at HidSharp.Reports.ReportDescriptor.ParseEncodedItem(EncodedItem item)
   at HidSharp.Reports.ReportDescriptor.ParseEncodedItems(IEnumerable`1 items)
   at HidSharp.Reports.ReportDescriptor.ParseRawReportDescriptor(Byte[] buffer)
   at HidSharp.Reports.ReportDescriptor..ctor(Byte[] buffer)
   at HidSharp.Platform.Linux.LinuxHidDevice.TryParseReportDescriptor(ReportDescriptor& parser, Byte[]& reportDescriptor)
   at HidSharp.Platform.Linux.LinuxHidDevice.RequiresGetInfo()
   at HidSharp.Platform.Linux.LinuxHidDevice.GetMaxInputReportLength()
   at OpenTabletDriver.Devices.HidSharpBackend.HidSharpEndpoint.<>c.<get_InputReportLength>b__7_0(HidDevice d)
   at OpenTabletDriver.Extensions.TryGet[TSource,TValue](TSource source, Func`2 predicate, TValue& value)
[Detect:Info]	No tablets were detected.
```

I've also tried this in my gentoo build, OTDd detected the tablet, the client showed the tablet area,
but after moving the cursor, the daemon throwed an error, most likely because I did not compile tablet support in the kernel(the tablet doesn't affect X11 with or without OTD). I will also compile tablet support later and report what would happen.

TheRealOne78, 2022-10-27:
I can also see there is a discord server that is to be preffered for this issue, shall we move this to discord?

jamesbt365, 2022-10-27:
That would speed up the process.

jamesbt365, 2022-10-27:
Also going to clear some things up.

You have the hid_uclogic module loaded, which is what otd is complaining at, tablet support can be off because its the kernel modules. What does need to be there is `/dev/uinput` and other things that you'll have basically.

you will also not be able to use that tablet configuration, that is why its crashing.

TheRealOne78, 2022-10-27:
Alright, made a new thread in Development #configs: https://discord.com/channels/615607687467761684/1035126135295582218/1035126138235789312

jamesbt365, 2024-03-26:
I can pick this back up after #3245 or my other PR for this is merged, until then this will remain how it is.