OpenTabletDriver/OpenTabletDriver #1480: CTE-430 broken

JJTech0130, 2021-10-04:
## Description

With recent development versions, support for the Wacom CTE-430 appears to be broken. If you go into the tablet debug menu, it shows that the report is not being parsed properly (or at all). Furthermore, if you try to use the eraser, the application will crash. This is not an issue in the stable release, however, I have anther tablet only supported by the beta and I would like to use it.
I have not tested this theory, but I wonder if https://github.com/OpenTabletDriver/OpenTabletDriver/commit/8f2a3eb6cf9ea7c6366a4697bdfff50bfd84e0f9 might have caused the issue?
## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System         | Windows 10
| OpenTabletDriver Version | https://github.com/OpenTabletDriver/OpenTabletDriver/commit/65cdcedf8cc0e9348e9e1ba13c4ef7bc019b057b
| Tablet                   | Wacom CTE-430 Sapphire

jamesbt365, 2021-10-04:
Can you export a diag for me

help > export diag

then save the file and send it here

JJTech0130, 2021-10-04:
I have attached the requested diagnostics.
[8f2a3eb-CTE-430.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7280997/8f2a3eb-CTE-430.zip)

JJTech0130, 2021-10-04:
The daemon crashes with this error whenever the eraser is used:
```
[IndexOutOfRangeException:Error]        Index was outside the bounds of the array.
   at OpenTabletDriver.Vendors.Wacom.Intuos.IntuosTabletReport..ctor(Byte[] report)
   at OpenTabletDriver.Vendors.Wacom.Intuos.IntuosReportParser.GetToolReport(Byte[] report)
   at OpenTabletDriver.Vendors.Wacom.Intuos.IntuosReportParser.Parse(Byte[] report)
   at OpenTabletDriver.Devices.DeviceReader`1.Main()
```

JJTech0130, 2021-10-04:
I need to test this a bit more, but I may have found the culprit: https://github.com/OpenTabletDriver/OpenTabletDriver/commit/755cab3db5685bbc4f2b9e772a9f96b46fbdc32a

JJTech0130, 2021-10-04:
Ok, so it seems like the parser for the CTE-430 is just plain broken. I'm not sure if it is just not used very often, but whatever.
I see 2 possible ways forward:
1) Revert the CTE-430 to the generic TabletReportParser, at the expense of some more advanced features
2) Write a new parser/modify the existing one

I would like some clarification about some things:
+ Does the Intuos parser work for other tablet devices?
+ Why was mouse support never implemented? I have a mouse for mine, and the driver can't tell the difference (It just treats it like a pen)

jamesbt365, 2021-10-04:
This is all stuff that has to be looked into. Modifying the parser so it doesnt break shouldnt take too long for someone who knows what they are doing. Thank you for this information so far.

I think mouse support is on the way some way or another and should be ready soon:tm:

JJTech0130, 2021-10-05:
Ok! Should I work on it? How soon is the eta? I only have the 430, so if I fix it then someone else would need to confirm it works for others.

With some trial and error, I did manage to get it (basically) working, and the mouse support can probably be copied from the V3 parser.

JJTech0130, 2021-10-05:
Is it just the 430 which is broken? Or is the Intuos parser broken? How does the numbering scheme work (I.e why is there Intuos and InuosV1 and Intuos3?) Should I create a new parser (IntuosLegacy /Intuos430 or something) just for the 430?

JJTech0130, 2021-10-05:
https://github.com/JJTech0130/OpenTabletDriver/tree/intuos-cte-430

jamesbt365, 2021-10-05:
If you have done testing and confirm it fixes it feel free to submit a pr.

jamesbt365, 2021-10-05:
If you think there is more you could add or you need further testing you could either wait to submit or submit as a draft anyway. 

If you need any other information that can be given to you then we are here.

gonX, 2021-12-06:
This is likely fixed by #1698. Could you test that for me please?

jamesbt365, 2021-12-31:
Closing due to the pr already being verified and lack of activity within the issue, this should be solved now.

closed 2021-12-31 by jamesbt365