OpenTabletDriver/OpenTabletDriver #1482: Support Request: Can You Change Pen Detection Height?

NeonScorpion, 2021-10-05:
# Support Request
Hello, I use this in Relative mode for playing FPS games. I realized that I have to lift my pen a lot higher than is comfortable for the tablet to stop detecting it and let me move my arm without moving the mouse. This causes issues where when aiming quickly if I just barely undershoot the height when I go to lift my arm and move it quickly I instead turn my entire camera around since the pen is still being detected. Is there any sort of setting I can adjust to lower or customize the distance from the Tablet the Pen has to be, to be detected? :)

I use a Wacom CTL-4100LW

## What I've already tried
I read the Configuration guide, but I don't really understand the lingo, so I don't want to mess with any settings and cause an issue.

## Diagnostic Information

jamesbt365, 2021-10-05:
Hover distance limiter

jamesbt365, 2021-10-05:
its a plugin

Mrcubix, 2021-10-05:
So here is the issue:

- on one side, we would had suggested to use hover distance, which is only compatible with some dev build, (not master cause ghost broke plugins and hasn't bothered to look into any of it yet)
- on the other side, these devs builds are filled with bugs, no because it's a dev build but cause a lot of breaking changes were made and most of the bugs resulting from these changes haven't been fixed yet (including 95% of of plugins being broken)

And if you need VmultiMode for a game such as Valorant then i don't know myself if it's not broken in the only dev builds where hover limiter work.

So here is 2 choice:

- You can try a specific dev build that will probably be linked by Kuuube
- You'll need to get used to it somehow

<sub><sup>skill issue</sub></sup>

>

NeonScorpion, 2021-10-05:
Hey thanks for the help. I'm new to this whole thing and using 3rd party Drivers. I will just get used to it, its not a super big deal, just thought there might be some sort of setting I missed. :)

ᵀʰᵃⁿᵏˢ ᶠᵒʳ ᵗʰᵉ ʰᵉˡᵖ

closed 2021-10-05 by NeonScorpion