OpenTabletDriver/OpenTabletDriver #1498: if you make otd detect tablet too quickly it crashes and the ux can lock up no matter if you close and reopen it

jamesbt365, 2021-10-10:
## Description

If you make otd detect too quickly through either the menu or by spamming control + d then otd freezes and crashes, upon reloading the ux doesn't respond at all, no clicks are registered even though the ux hasn't crashed.

[image]

It gets stuck on the select tablet and mode but otd doesn't work even if clicked.

## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System   Linux      |
| OpenTabletDriver Version 0.6 pre release (arch aur the day before the config folder move)|
| Tablet         XP PEN Star G640 Rev A          |

jamesbt365, 2021-10-31:
This seems to be some logging issue, it detection doesnt need to fast. Shown here:
https://streamable.com/cfbfs6

Sublimelime, 2021-10-31:
[Putting this here for documentation and record-keeping]

Currently testing by commenting out line [here](https://github.com/OpenTabletDriver/OpenTabletDriver/blob/v0.5.3.3/OpenTabletDriver/Driver.cs#L91) -- Issue is apparently fixed off that, and will print only the "found tablet" messages with a bit of delay (queuing effect is obvious) and never crash no matter how hard you spam.

Issue may be related to heavy spam of info messages - Testing a compromise fix by downgrading "Searching" messages to Debug level so they don't show by default.

Sublimelime, 2021-10-31:
Some other misc things noticed:

- GUI cannot be reopened after crashing this way - crash seems to bork something in communication between the daemon and GUI, though Daemon itself seems fine (tablet still works, daemon still prints messages to console) Gotta restart both to fix.
- Attempting to wrap different parts of Driver.Detect in lock statements doesn't appear to have an effect, only thing that has worked is commenting out that Log.Write.

We can either batch these log statements to avoid spamming somehow, or just not log searching statements (only "found" and "not found") or try and see if we can figure out why spamming logs crashes the gui.

InfinityGhost, 2021-11-11:
Resolved in #1603

closed 2021-11-11 by InfinityGhost