OpenTabletDriver/OpenTabletDriver #1525: Support Request: Detecting a Gaomon tablet

Puppetted, 2021-10-16:
# Support Request
I want to use OpenTabletDriver for my Gaomon 1060pro, but after all that I have tried, I can't get it to work

## What I've already tried
I have used WinUSB to do cleanup.
## Diagnostic Information

jamesbt365, 2021-10-16:
I dont have any diagnostic information to base this off, go to help > export diagnosis, then save that file and send it here.

Puppetted, 2021-10-16:
[Diagnostics.r.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7357342/Diagnostics.r.zip)
here you go.

jamesbt365, 2021-12-28:
I haven't seen gaomon use this pid before, are you 100% sure that other drivers are not installed?

Puppetted, 2021-12-29:
I'm quite sure. Outside of the Gaomon driver and otd, I've never attempted to download any tablet driver. And I always make sure to run the WinUSB script to delete the Gaomon one. If there were other drivers on my computer, it may be from automatic Windows update and I have no way to know where they are.

jamesbt365, 2021-12-29:
Can you give me another diagnostic please?

Puppetted, 2021-12-29:
[Diagnostic.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7788807/Diagnostic.zip)
Here you go

jamesbt365, 2021-12-29:
go to tablet > device string reader, 9580 top box and 100 in the middle box, then click dump all. Send that file here.

Puppetted, 2021-12-29:
Where is that file, I input the number and click dump all but there is no popup telling me that I've made a file. Is it the diagnostics again?

jamesbt365, 2021-12-29:
There should be a pop up telling you to save a file if you click dump all.

Puppetted, 2021-12-29:
[dump.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7788830/dump.txt)
Found it. I need to input a random string number as well it seems.

jamesbt365, 2021-12-29:
Try using this config, [1060pro.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7788868/1060pro.zip) Put this in Configurations\Gaomon and replace the old one

Puppetted, 2021-12-29:
IT WORKS!!!!! It says connected now. Thank you very much!

closed 2021-12-29 by Puppetted

jamesbt365, 2021-12-29:
does all of the features work on it?

Puppetted, 2021-12-29:
Yes. I can edit the tablet area to suit my need. Is there a template feature? It is very convenient to switch between the full area noting mode and tini area osu mode.

jamesbt365, 2021-12-29:
You can export settings into a file and load them, in `master`, this have been sped up slightly but master is unstable so i don't recommend using it as plugins haven't been completely updated yet and a few other bugs exist there.