OpenTabletDriver/OpenTabletDriver #1529: Auxillary Binds on XP-PEN Deco 03 not working

RedCyro, 2021-10-17:
so i installed this about a month ago and have had this problem and while its not too big of an issue its pretty inconvenient.

Whenever i bind something to either the tablet, or the pens auxillary buttons, they don't work when i press the button, i've reinstalled the drivers several times, but to no avail.

I cant think of much more information to give, so please ask questions if needed

gonX, 2021-10-17:
There's a chance OTD isn't parsing auxiliary buttons for that specific tablet yet. Do the auxiliary buttons show up in the tablet debugger? (Help -> Tablet Debugger in OpenTabletDriver)
They should be in the Tablet Report field if they exist

If they don't (or if they show incorrectly), a change is needed in the parser which is usually a fairly simple fix.

Please specify operating system (and its version) as well as the OpenTabletDriver version you're using, as that will help us narrow down how to help you best.

gonX, 2021-12-07:
After further investigation, this tablet is only using the generic `TabletReportParser` in 0.5.3.1 which has no support for aux buttons, and there's a chance pen buttons might not be parsed correctly with this parser.

You'd have to use [Mrcubix' backport plugin](https://github.com/Mrcubix/OTD.Backport/), or use a pre-release of OpenTabletDriver to get auxiliary button functionality working.

gonX, 2021-12-20:
Closing stale support request

closed 2021-12-20 by gonX