OpenTabletDriver/OpenTabletDriver #1530: Support Request: Pressure suppot (Windows Ink) "NullReferenceException"

Wine-Bottle, 2021-10-17:
# Support Request

- My objective is to enable pen pressure support, so I can use it in Photoshop
After installing Windows Ink and VMulti, and changing the tip bindings to "Windows Ink: Pen Tip", using the pen tip will cause OTD to undetect my tablet (Wacom CTL-680)

## What I've already tried
- Deleting OTD and using a refresh download
- Remove VMulti and Windows Ink, and reinstalling them

## Diagnostic Information

Console output (I removed the irrelevant "searching for *" other models)

  [Detect:Info]	Found tablet 'Wacom CTL-680'
  [Settings:Info]	Output mode: Relative Mode
  [Settings:Info]	Relative Mode Sensitivity (X, Y): <10, 10>
  [Settings:Info]	Relative Mode Rotation: 0
  [Settings:Info]	Reset time: 00:00:00.1000000
  [Settings:Info]	Tip Binding: [VoiDPlugins.OutputMode.WinInkButtonHandler]@5%
  [Settings:Info]	Pen Bindings: [0, VoiDPlugins.OutputMode.WinInkButtonHandler], [1, VoiDPlugins.OutputMode.WinInkButtonHandler]
  [Settings:Info]	Express Key Bindings: [0, ], [1, ], [2, ], [3, ], [4, ], [5, ], [6, ], [7, ]
  [Settings:Info]	Driver is auto-enabled.
  ``Pen tip touching the tablet``
  [NullReferenceException:Error] Object reference not set to an instance of an object.

*.json is not supported, I renamed the file extension to .txt.
[20211018_diag.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7360483/20211018_diag.txt)

jamesbt365, 2021-10-17:
Have you got vmulti driver, this isnt vmulti mode btw.
https://github.com/X9VoiD/vmulti-bin/releases/tag/v1.0

Wine-Bottle, 2021-10-18:
> 
> 
> @Mrcubix you forgot to mention to switch output mode to `Windows Ink`, but yeah the issue was that the user forgot to use the windows ink output mode as shown by `Output mode: Relative Mode`.

Yea, because after switching and when the pen tip touches the tablet, it throws null exception, it didn't get the chance to do anything before and afterwards