OpenTabletDriver/OpenTabletDriver #1533: Mouse pointer no longer can interact with the computer if you lift the stylus away from the surface of the tablet while holding down a button on the stylus

K4LCIFER, 2021-10-18:
## Description
If you touch the stylus to the surface of the tablet, hold down one of the stylus's buttons, and then lift the stylus away from the surface of the tablet (while keeping the button pressed down), until the stylus is out of range of the tablet, you can observe that the mouse pointer will no longer work (as in it will move around, but it will not be able to click on anything/interact with anything) until the stylus is brought back into range of the tablet without the button pressed down.

## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System         | Arch Linux |
| OpenTabletDriver Version | 0.5.3.3-1|
| Tablet                   | XP-Pen Artist Pro 12|

Sublimelime, 2021-10-18:
For additional clarification:

- What happens if you bring the stylus within range again while still holding the pen button? Does it still work fine? What if you keep holding as you bring it closer, then release the pen button while in range?
- Does unplugging the tablet after causing the bug allow the mouse to work again?
- Does re-detecting the tablet via the menu allow the mouse to work again?

K4LCIFER, 2021-10-18:
@Sublimelime unfortunately, `<ctrl> + d` interferes with other keybindings for me.

K4LCIFER, 2021-10-18:
> this has somehow a deep smell of hardware issue, but just to make sure, check if the second byte change in tablet debugger when your pen goes out of range with one of the buttons held

The second byte changes from `A2` to `C0` when the pen moves out of range.

gonX, 2021-12-01:
#1676 should handle this for your tablet if you use Linux Artist Mode.