OpenTabletDriver/OpenTabletDriver #1573: Allow overrides on parsers

Mrcubix, 2021-10-30:
## Description

I think people shouldn't have to download the pre-release or master's versions just to use their tablets with OTD.
Just like configurations, we should be able to add overrides so people can download parsers around in whatever proper form is suitable for the project.
A solution similar to plugins in the past should be used, but because it was changed and broken by ghost (no it's still not fixed), it will probably require some work on that side before considering this feature.

jamesbt365, 2021-10-30:
This feature would be really useful for not having to send a build to somebody each time a parser has been changed, instead allowing you to just give them a parser to swap out.

X9VoiD, 2021-10-30:
If you meant *"override"* as in *"change the behaviour of already working parsers"* then it's not an acceptable feature to add. It only adds extra complexity in development cycles just to skip building OTD. Please remember, that there are other projects that rely on `OTD.Core` and doesn't have any plugin support.

Overridden parsers *may* be able to fix some issues on existing parsers but then they will have to be merged to OTD so that the fix becomes available to other projects, basically making the overrides obsolete the moment OTD merges the fix.

If you meant *"as a loadable plugin"*, then we already have it. `PluginManager` loads everything that implements an interface or abstract class from `OpenTabletDriver.Plugin` namespace. This includes `IReportParser<>`.

Will close since in either of the situations above it'll either be useless, or we already have it.

closed 2021-10-30 by X9VoiD