OpenTabletDriver/OpenTabletDriver #1575: Windows ink/VMulti broken in 0.5.3+

Doksha, 2021-10-30:
## Description

Vmulti plugins (Windows ink and VMulti mode I know for sure) do not work in 0.5.3+
Also tested with an alternate computer and got the same results.
## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System         | Windows 10
| OpenTabletDriver Version | 0.5.3
| Tablet                   | Veikk A15

Mrcubix, 2021-10-30:
you didn't install the [requirement](https://github.com/X9VoiD/vmulti-bin/releases/latest), re-open the issue if still not fixed

Doksha, 2021-10-30:
Plugin is fully up to date, also yes of course I installed the requirement. (note that it does work in 5.2.3 and below)

jamesbt365, 2021-10-30:
Export a diag help > export diagnosis

Doksha, 2021-10-30:
Alright, apparently the vmulti tip button selection is a little buggy and just needed to be selected multiple times (although it should be the default selection since mouse button does not work with vmulti)

Mrcubix, 2021-10-30:
it's very probable then that the value for the tip binding just changed, and you updated that value when you selected it again and saved

InfinityGhost, 2021-11-01:
Appears resolved from conversation.

closed 2021-11-01 by InfinityGhost