OpenTabletDriver/OpenTabletDriver #1579: Linux Artist Mode incorrectly tells libinput about tool proximity on tablets that don't implement IProximityReport

gonX, 2021-10-31:
## Description
How to test:

Firstly make sure you're running in Linux Artist Mode and you're using a tablet that does not implement `IProximityReport`s (check your parser)

1. Run `libinput debug-tablet`
2. Bring a pen in range, then take it out of range again.
3. Observe that `BTN_TOOL_PEN` or `BTN_TOOL_RUBBER` buttons remains active despite being out of range.

If no buttons show up, restart OTD and make sure `libinput debug-tablet` is running before bringing the pen in range.

## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System         | Linux
| OpenTabletDriver Version | 267a322
| Tablet                   | XP-Pen Star G960S Plus

Sublimelime, 2021-10-31:
Seems related to issue #1533

gonX, 2021-10-31:
It's not the same, but it is similar in that they're hardware detects that we have to work around.

XP Pen does have out of range detection (look for 0xC0 in `report[1]`), so we likely just need to reimplement proximity and/or split `IProximityReport` into 2 separate things, `IInRangeReport` and `IDistanceReport`  (or similar naming), so we can set InProximity as long as we're not encountering 0xC0.

gonX, 2021-11-30:
#1676 handles this for the tablets that are covered in the PR (currently intuosv1 and xppen parser). All parsers needs to implement `OutOfRangeReport` (assuming the PR is accepted) to close this bug successfully, or we need to synthesize an `OutOfRangeReport` for tablets that do not have `OutOfRangeReport`s implemented - it's best we don't synthesize them on tablets that have them.

gonX, 2021-12-23:
If this can't be solved properly in time for 0.6.x I am also willing to look at alternatives where we simply add a timer to Linux Artist Mode which does the reset when report rate drops below a certain threshold.

Ideally the output mode should be aware of whether a tablet/parser can report this in the first place - if this isn't accounted for we at worst lift buttons (and pressure) when we don't intend to.

InfinityGhost, 2021-12-23:
> If this can't be solved properly in time for 0.6.x I am also willing to look at alternatives where we simply add a timer to Linux Artist Mode which does the reset when report rate drops below a certain threshold.

We recently added a report type to handle out of range reports. This approach could be used to handle that case, however it would be difficult to determine if a tablet actually reports it (except by the report parser, which still isn't a reliable way to do it). Best option IMO is to always assume in range (for now) for the tablets that don't support it and properly do it for those who can. 

> If this is the desired approach, ideally the output mode should be aware of whether a tablet/parser can report this in the first place - if this isn't accounted for we at worst lift buttons (and pressure) when we don't intend to.

The parsers do not have a reliable way to determine what features may be present, nor if the tablet in which is using it will properly use those features.

gonX, 2021-12-23:
No, but we could change `OutOfRangeReport` to be `IOutOfRange` and add a field that tells whether the parser will send it. We'd just have to replace `if is OutOfRangeReport` checks with `if is IOutOfRange && report.IsOutOfRange`.

gonX, 2021-12-23:
Btw
> Best option IMO is to always assume in range (for now) for the tablets that don't support it and properly do it for those who can.

I kinda agree and it effectively means we can probably drop this from 0.6.x milestone. It's less breaking than #1533 and fixing that most likely fixes this, especially if we go with complete `OutOfRange` reporting

InfinityGhost, 2021-12-23:
Let's go that route then. The less we have on the plate the better for now.