OpenTabletDriver/OpenTabletDriver #1580: Reinstalling same plugin from file crashes UX

Sublimelime, 2021-10-31:
## Description
Installing a plugin from a .dll, uninstalling then reinstalling from that same file crashes the UX. Daemon remains okay and UX can be relaunched, though the reinstall is not successful. **Both daemon and UX must be restarted to successfully reinstall plugin.**

## Reproduction

- Install plugin from Plugin manager -> file -> install plugin
- Remove plugin by clicking on plugin in list -> Uninstall plugin
- Reinstall plugin from same file, via same method as above
- Crash

## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System         | GNU/Linux Fedora 34
| OpenTabletDriver Version | Master(267a32282ad5462a5be254e74f60edeba809831f) - Compiled from source
| Tablet                   | Gaomon S620

Crash stacktrace:

[stacktrace.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7449853/stacktrace.txt)

File used (though should be plugin agnostic): [Updated version of Kuuube chatter exterm.](https://cdn.discordapp.com/attachments/628651971267788800/884592400402632744/Kuuubes_CHATTER_EXTERMINATOR.dll)

X9VoiD, 2021-10-31:
Looks very similar to #720. The UX crash seems to be distro-specific.

Sublimelime, 2021-11-15:
Researching this issue - current lead is a certain line of the stacktrace -

`StreamJsonRpc.RemoteInvocationException: The file '/run/user/1000/OpenTabletDriver/Kuuubes_CHATTER_EXTERMINATOR/Kuuubes_CHATTER_EXTERMINATOR.dll' already exists.`

`/run/user/1000` is [basically a temporary directory for files](https://unix.stackexchange.com/a/162911) owned by processes - it looks like OTD or the RPC or something doesn't properly clear out this file when it uninstalls a plugin.

Gonna look into it deeper tomorrow and see if I can at best completely fix, at worst stop it from crashing.

gonX, 2022-11-07:
Need retest please

Mrcubix, 2025-09-29:
This needs to be retested now that a lot of changes related to plugin loading & unloading were made.

closed 2025-10-05 by gonX