OpenTabletDriver/OpenTabletDriver #1598: windows 11 cursor 30 fps bug

vetyalmao, 2021-11-03:
i updated to windows 11, and when i try to play with my tablet, it moves my cursor in 30 fps. not just in osu. to fix this, i need to minimize every single open application.

jamesbt365, 2021-11-03:
Check in the debugger for report rate, go into tablet > tablet debugger and check the report rate  while moving

Mrcubix, 2021-11-03:
I mean, there aren't anything to be done on our side.
Windows 11 is already a terrible OS, and i'm pretty sure your windows install is messed up in some way, as most other win 11 OTD users don't have issues.

vetyalmao, 2021-11-03:
> Check in the debugger for report rate, go into tablet > tablet debugger and check the report rate while moving

when i lift up the pen it showed this. while moving i saw 132-133
[image]

vetyalmao, 2021-11-03:
> I mean, there aren't anything to be done on our side. Windows 11 is already a terrible OS, and i'm pretty sure your windows install is messed up in some way, as most other win 11 OTD users don't have issues.

wtf do you mean "your windows install is messed up in some way" you cant fuck up a windows installation. jesusomega uses windows 11, otd and doesnt have this problem. ive had this problem on windows 10, on another pc, another tablet.

jamesbt365, 2021-11-03:
Oh please shush about you fucking up a windows install, it is completely possible because we have seen it countless times, stop being arrogant and give us actual fucking proof about your claim before you shout shit at us.

record a fucking video or something before you start throwing false info at us.

vetyalmao, 2021-11-03:
> Oh please shush about you fucking up a windows install, it is completely possible because we have seen it countless times, stop being arrogant and give us actual fucking proof about your claim [...]

sowwy

jamesbt365, 2021-11-03:
Seems to be resolved in discord.
https://canary.discord.com/channels/615607687467761684/615611007951306863/905576247315271731

Closing issue now.

closed 2021-11-03 by jamesbt365