OpenTabletDriver/OpenTabletDriver #1600: Parblo A610Pro

jaxxtrend, 2021-11-03:
# Tablet Information

| Name            | Value |
| --------------- | ----- |
| Manufacturer    |  Parblo
| Product Name    | A610 Pro
| Width           | 1005
| Height          | 629
| Pressure Levels | 8192
| LPI             | 5080LPI

## Diagnostic Information

[Diag.log](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7470259/Diag.log)

jamesbt365, 2021-11-03:
Extract this file into your configurations folder based where you exectuables are.

[A610.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7470307/A610.zip)

Get back to me when you are done. And go into the tablet > tablet debugger

jaxxtrend, 2021-11-03:
with config pen is detected and i see pen pressure in tablet debugger, but pressure not work in ps and krita in both modes (abs or relative)
[diag2.log](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7470663/diag2.log) diag with config

jamesbt365, 2021-11-03:
well, because the config isn't completed we can't get to that yet, in the debugger can you get the max x and y coords by going to the bottom right of your tablet slowly. I need the highest you can get.

jaxxtrend, 2021-11-03:
okay, this is my records 51080 31991

jamesbt365, 2021-11-03:
Try this one: can you reach all the corners of your screen using full area?, can you go back into the debugger and tell me if the pen buttons read on there for me.
[A610.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7470760/A610.zip)

jaxxtrend, 2021-11-03:
I tested and i think better values equal is 51100 and 32000 with this values result is better. Button on the pen and tablet in debugger work fine(False changes to True) excluded ring and button in center of ring on the tablet (not have this buttons in debug)

jamesbt365, 2021-11-03:
Ok, i can correct that for you now and you give me another test on position, give me one moment.

jamesbt365, 2021-11-03:
Give this a go:
[A610.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7470830/A610.zip)

jaxxtrend, 2021-11-03:
Work fine, pen at the edge

jamesbt365, 2021-11-03:
How many tablet buttons does your tablet have?

Can you bind a few to something in the binding tab and give me a test?

jaxxtrend, 2021-11-03:
On the tablet have 8 standard buttons and additionally have the ring (work like mouse wheel) with button in center of ring. Pen have two buttons. How i can give the test result to you?

jamesbt365, 2021-11-03:
Can you just test binding a few buttons in the bindings tab of otd and testing if they work? I just need to know if they work when bound.

jaxxtrend, 2021-11-03:
Yes, work fine

jamesbt365, 2021-11-03:
When in the debugger, when using the ring, what bytes (1 set of 2 numbers) change?

And when pressing the button on the ring also get me this information.

jaxxtrend, 2021-11-04:
Raw tablet Data:
if wheel button not pressed:
02 F0 00 00 00 00 00 00 00 00

if wheel button pressed:
02 F0 00 01 00 00 00 00 00 00

and if wheel rotate clockwise:
02 F0 00 00 00 00 00 01 00 00

and if wheel rotate counter-clockwise:
02 F0 00 00 00 00 00 02 00 00

jamesbt365, 2021-11-04:
Can you get me the data for the normal tablet buttons not counting these ones please?

jaxxtrend, 2021-11-04:
From up to down Changes only this set 02 F0 **00** 00 00 00 00 00 00 00
1. 01
2. 02
3. 04
4. 08
5. 10
6. 20
7. 40
8. 80

jaxxtrend, 2021-11-04:
Also this tablet have the pen tilt this feature not supported too?

jaxxtrend, 2021-11-04:
I'll ask directly how much it will cost to add wheel support, pen tilt and pen pressure in Photoshop or/and Krita in relative mode.

jamesbt365, 2021-11-04:
This is a free service, and this will get done no matter if you decide to donate or not.

For the next step can you download the pre release of OpenTabletDriver for me, you can get it here
https://cdn.discordapp.com/attachments/615939905566015493/890625738691211334/OpenTabletDriver.win-x64.zip

Close your other build of OpenTabletDriver.
Extract this to its own directory that isn't related to the other one and create a `userdata` folder.

Then download [A610Pro.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7477506/A610Pro.zip) and extract it into the configurations folder based with the new configurations folder (the old config won't work with the new build)

Then do Tablet > Detect Tablet.

Notify me once you are there.

jaxxtrend, 2021-11-04:
I am ready.

jamesbt365, 2021-11-04:
Turn on enable data recording in the Tablet > Tablet Debugger and put your pen on the tablet, from here tilt your pen around in a circle. do a circle 2-3 times and stop the data recording. Then send the file here. It should be in the `userdata` folder.

jaxxtrend, 2021-11-04:
[tablet-data.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7478092/tablet-data.txt)

jamesbt365, 2021-11-04:
Try out this config please
[A610Pro.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7478273/A610Pro.zip)
:

jaxxtrend, 2021-11-04:
When I touch the tablet with the pen, the user interface loses detection of the tablet

jamesbt365, 2021-11-04:
Export a diag when it looses connection to the tablet

Help > export diagnosis

jaxxtrend, 2021-11-04:
[diag3.log](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7478340/diag3.log)

jamesbt365, 2021-11-04:
Use this config again, is there a tilt debug value?

[A610Pro.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7478396/A610Pro.zip)

jaxxtrend, 2021-11-04:
Yes, tilt values are present

jamesbt365, 2021-11-04:
Can you give everything a full test, you won't be able to use the wheel or pressure as pressure isn't supported on the pre releases and the wheel is currently not able to be supported.

So test pressure, tilt, buttons, whether or not you can reach all the corners, Can you confirm all of that for me.

closed 2021-11-05 by X9VoiD

jaxxtrend, 2021-11-05:
> 
> 
> Can you give everything a full test, you won't be able to use the wheel or pressure as pressure isn't supported on the pre releases and the wheel is currently not able to be supported. However pressure is supported in standard releases with the use of a plugin in the plugin manager. (Follow the wiki)
> 
> So test pressure, tilt, buttons, whether or not you can reach all the corners, Can you confirm all of that for me.

Buttons, except wheel, work fine. Corners reached. In debug tilt and pressure work fine.
Have eraser button in debug and pen settings, but this pen tablet no have eraser button.
Сan you tell how long it will take to support the wheel?

X9VoiD, 2021-11-05:
The binaries here have support for the wheel button: https://github.com/OpenTabletDriver/OpenTabletDriver/actions/runs/1423797301

Support for the wheel itself cannot be added yet since it is still in development and not yet implemented in OTD's core functionality. See #1367 for tracking the progress on wheel.