OpenTabletDriver/OpenTabletDriver #1603: UI deadlock from logging too rapidly

InfinityGhost, 2021-11-04:
## Description

UI deadlocks attempting to write to the log asynchronously. Reproducable by replugging a device rapidly.

Stack trace from attached debugger paused at deadlock
```
SemaphoreSlim.WaitUntilCountOrTimeout()
SemaphoreSlim.Wait()
BlockingCollection<LogMessage>.TryAddWithNoTimeValidation()
BlockingCollection<__Canon>.Add()
LogDataStore.Add()
LogView.<>c__DisplayClass4_0.<AddMessage>b__0()
ApplicationHandler.<>c__DisplayClass20_0.<AsyncInvoke>b__0()
Application.InvokeCB.Invoke()
Timeout.TimeoutProxy.Handler()
[Managed to Native Transition]
Application.Run()
ApplicationHandler.Run()
Application.Run()
App.Run()
Program.Main()
```

## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System         | NixOS 21.11
| OpenTabletDriver Version | cf53686ee9077d5b4fab753d55b4e48a66e19324
| Tablet                   | CTL-470

Sublimelime, 2021-11-04:
This also appears to be the cause of #1498 (just throwing this here to link them together).

InfinityGhost, 2021-11-11:
Fixed in #1619

closed 2021-11-11 by InfinityGhost