OpenTabletDriver/OpenTabletDriver #1605: Plugin Manager

Killua2134, 2021-11-04:
Hello my plugin manger does not open can you help me fix it thank you

jjhhgg100123, 2021-11-04:
Please provide basic diagnostic information, including but not limited to OS Version, OTD version, and exported diagnostics. The template exists for a reason.

Killua2134, 2021-11-04:
Its enabled but still has the warning

coinworld, 2021-11-04:
You are probably using Administrator account, which a normal user shouldn't use. (I think you are using debloated OS)

You can try making a normal account and run OTD in it, but you'd better use normal OS if you are using debloated one.

X9VoiD, 2021-11-04:
Duplicate of #681

closed 2021-11-04 by X9VoiD

creidhne-a16i, 2024-07-03:
Not to be an ass, but using debloated OS is NOT better. Why wouldn't the plugin window open at all just because the user has higher privileges than normal? It's not like you're limited on permissions in that situation.

jamesbt365, 2024-07-03:
It is an upstream bug with the upstream UI framework we use, this not working is not a personal choice.

creidhne-a16i, 2024-07-10:
Here's how I see this: yes, I'm using debloated windows, and yes I'm running it with a single Administrator account. I get it that it's an upstream problem, and not something you can fix; and that I shouldn't be running as admin. But... at the same time, in the 5+ years that I've been running this exact system as it is now, not a single program had an issue like this. None. Please don't get me wrong, I absolutely love the idea of open source driver, and the functionality that it provides despite the somwhat involved setup. What I did is setup a temporary account just to get the plugin manager to open, edit what I needed, and go back to Admin account and run it there - and it has been fine... until I needed the mouse clicks to work, which required more plugins, at which point I decided to revert back to wacom driver. Why let the window-opening issue like that just sit untouched and not do some kind of workaround? Why not switch to some other framework that doesn't cause this kind of issues?

gonX, 2024-07-10:
@creidhne-a16i 
We're just as frustrated with the issue as you are. We haven't been able to figure out any workarounds yet, and a GUI framework change is in the works but is taking longer than expected (#2772). It's solving a lot of our issues which is why nobody has bothered looking deep into a workaround