OpenTabletDriver/OpenTabletDriver #1607: The script for "The driver fails to open the tablet (DeviceIOException)" no longer works.

ToxicGLaDOS, 2021-11-05:
## Description

When running the provided script [here](https://github.com/OpenTabletDriver/OpenTabletDriver/wiki/Linux-FAQ#the-driver-fails-to-open-the-tablet-deviceioexception)
```
git clone https://github.com/InfinityGhost/OpenTabletDriver-udev.git
cd OpenTabletDriver-udev
git submodule update --init --remote
./build.sh
...
```
it fails with
```
Unhandled exception: System.IO.DirectoryNotFoundException: Could not find a part of the path '/home/me/projects/third-party/OpenTabletDriver-udev/.modules/OpenTabletDriver/OpenTabletDriver/Configurations'
```
That folder no longer exists. As a workaround I went into `.modules/OpenTabletDriver` and did a `git checkout v0.5.3.3`. Looks like it was removed in [d8640acfd13a1eea71628fdc0b79646eed207417](https://github.com/OpenTabletDriver/OpenTabletDriver/commit/d8640acfd13a1eea71628fdc0b79646eed207417).

After some more investigating I think the issue is with how the `--remote` pulls in newer changes. `--remote` makes the submodule use the master branch of OpenTabletDriver instead of [9ed8069](https://github.com/OpenTabletDriver/OpenTabletDriver/tree/9ed806996b66fff557d9491f3a155a960c2768b7) as it's linked to right now in [OpenTabletDriver-udev](https://github.com/InfinityGhost/OpenTabletDriver-udev/tree/master/.modules).

tl;dr
We should remove `--remote` from the script in [this section](https://github.com/OpenTabletDriver/OpenTabletDriver/wiki/Linux-FAQ#the-driver-fails-to-open-the-tablet-deviceioexception) of the wiki.

## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System         | Linux
| OpenTabletDriver Version | master
| Tablet                   | n/a

X9VoiD, 2021-11-05:
Wiki says:

[image]

"here" links to: https://opentabletdriver.net/Wiki

But even that is a little bit outdated, so here is a working one:
```
git clone https://github.com/InfinityGhost/OpenTabletDriver-udev.git
cd OpenTabletDriver-udev
git submodule update --init --remote
./build.sh
sudo mv ./bin/99-opentabletdriver.rules /etc/udev/rules.d/99-opentabletdriver.rules
sudo udevadm control --reload-rules && sudo udevadm trigger
cd ..
rm -rf OpenTabletDriver
rm -rf OpenTabletDriver-udev
```
Transferring issue to OTD.Web

InfinityGhost, 2021-11-05:
> But even that is a little bit outdated, requesting transfer to OTD.Web @InfinityGhost

I recently updated this in https://github.com/OpenTabletDriver/OpenTabletDriver.Web/pull/12, what else would be considered outdated @X9VoiD?

X9VoiD, 2021-11-05:
the .rule file is generated under `bin` directory, not `build`.

jamesbt365, 2021-11-06:
> > But even that is a little bit outdated, requesting transfer to OTD.Web @InfinityGhost
> 
> I recently updated this in [OpenTabletDriver/OpenTabletDriver.Web#12](https://github.com/OpenTabletDriver/OpenTabletDriver.Web/pull/12), what else would be considered outdated @X9VoiD?

Also putting to your attention that ArgumentOutOfRangeException the other solution blacklisting hid_uclogic hasn't been added yet. This solution also works where the other solution doesn't sometimes.

gonX, 2021-12-06:
Fixed, open a new issue in [OpenTabletDriver/OpenTabletDriver.Web](https://github.com/OpenTabletDriver/OpenTabletDriver.Web) if the new wiki is incorrect.

closed 2021-12-06 by gonX