OpenTabletDriver/OpenTabletDriver #1622: Support Request: H610PRO not detected, tried cleanup in FAQ

Coolykoen, 2021-11-10:
# Support Request
Tablet not detected, no other driver currently installed.

## What I've already tried
I did try the cleanup tool in the FAQ but it still won't detect the tablet.

Then i looked at other issues with detection and saw "tablet > device string reader" but that button doesn't do anything for me i think?

## Diagnostic Information
```
[Detect:Info]	Searching for tablet 'Gaomon 1060 Pro'
[Detect:Info]	Searching for tablet 'Gaomon M106K Pro'
[Detect:Info]	Searching for tablet 'Gaomon M106K'
[Detect:Info]	Searching for tablet 'Gaomon M10K'
[Detect:Info]	Searching for tablet 'Gaomon S56K'
[Detect:Info]	Searching for tablet 'Gaomon S620'
[Detect:Info]	Searching for tablet 'Huion 1060 Plus'
[Detect:Info]	Searching for tablet 'Huion G10T'
[Detect:Info]	Searching for tablet 'Huion GC610'
[Detect:Info]	Searching for tablet 'Huion H1060P'
[Detect:Info]	Searching for tablet 'Huion H1161'
[Detect:Info]	Searching for tablet 'Huion H430P'
[Detect:Info]	Searching for tablet 'Huion H610 Pro V2'
[Detect:Info]	Searching for tablet 'Huion H610 Pro'
[Detect:Info]	Searching for tablet 'Huion H640P'
[Detect:Info]	Searching for tablet 'Huion H950P'
[Detect:Info]	Searching for tablet 'Huion HC16'
[Detect:Info]	Searching for tablet 'Huion HS610'
[Detect:Info]	Searching for tablet 'Huion HS611'
[Detect:Info]	Searching for tablet 'Huion HS64'
[Detect:Info]	Searching for tablet 'Huion Kamvas 13'
[Detect:Info]	Searching for tablet 'Huion Kamvas 16'
[Detect:Info]	Searching for tablet 'Huion Kamvas Pro 13'
[Detect:Info]	Searching for tablet 'Huion New 1060 Plus'
[Detect:Info]	Searching for tablet 'Huion osu!tablet'
[Detect:Info]	Searching for tablet 'Huion Q11K V2'
[Detect:Info]	Searching for tablet 'Huion Q11K'
[Detect:Info]	Searching for tablet 'Huion Q620M'
[Detect:Info]	Searching for tablet 'Huion WH1409 V2'
[Detect:Info]	Searching for tablet 'VEIKK A15 Pro'
[Detect:Info]	Searching for tablet 'VEIKK A15'
[Detect:Info]	Searching for tablet 'VEIKK A30'
[Detect:Info]	Searching for tablet 'VEIKK A50'
[Detect:Info]	Searching for tablet 'VEIKK S640'
[Detect:Info]	Searching for tablet 'Wacom CTE-430'
[Detect:Info]	Searching for tablet 'Wacom CTE-440'
[Detect:Info]	Searching for tablet 'Wacom CTE-450'
[Detect:Info]	Searching for tablet 'Wacom CTE-460'
[Detect:Info]	Searching for tablet 'Wacom CTE-640'
[Detect:Info]	Searching for tablet 'Wacom CTE-650'
[Detect:Info]	Searching for tablet 'Wacom CTF-430'
[Detect:Info]	Searching for tablet 'Wacom CTH-460'
[Detect:Info]	Searching for tablet 'Wacom CTH-470'
[Detect:Info]	Searching for tablet 'Wacom CTH-480'
[Detect:Info]	Searching for tablet 'Wacom CTH-490'
[Detect:Info]	Searching for tablet 'Wacom CTH-670'
[Detect:Info]	Searching for tablet 'Wacom CTH-680'
[Detect:Info]	Searching for tablet 'Wacom CTH-690'
[Detect:Info]	Searching for tablet 'Wacom CTL-4100'
[Detect:Info]	Searching for tablet 'Wacom CTL-4100WL'
[Detect:Info]	Searching for tablet 'Wacom CTL-460'
[Detect:Info]	Searching for tablet 'Wacom CTL-470'
[Detect:Info]	Searching for tablet 'Wacom CTL-471'
[Detect:Info]	Searching for tablet 'Wacom CTL-472'
[Detect:Info]	Searching for tablet 'Wacom CTL-480'
[Detect:Info]	Searching for tablet 'Wacom CTL-490'
[Detect:Info]	Searching for tablet 'Wacom CTL-6100WL'
[Detect:Info]	Searching for tablet 'Wacom CTL-671'
[Detect:Info]	Searching for tablet 'Wacom CTL-672'
[Detect:Info]	Searching for tablet 'Wacom CTL-680'
[... 164 lines cut ...]
[Detect:Info]	Searching for tablet 'Wacom XD-1212-U'
[Detect:Info]	Searching for tablet 'Wacom XD-1218-U'
[Detect:Info]	Searching for tablet 'XP-Pen Artist 12 Pro'
[Detect:Info]	Searching for tablet 'XP-Pen Deco 01 V2'
[Detect:Info]	Searching for tablet 'XP-Pen Deco 01'
[Detect:Info]	Searching for tablet 'XP-Pen Deco 02'
[Detect:Info]	Searching for tablet 'XP-Pen Deco 03'
[Detect:Info]	Searching for tablet 'XP-Pen Deco Mini7'
[Detect:Info]	Searching for tablet 'XP-Pen Deco Pro Medium'
[Detect:Info]	Searching for tablet 'XP-Pen Deco Pro Small'
[Detect:Info]	Searching for tablet 'XP-Pen Star 03'
[Detect:Info]	Searching for tablet 'XP-Pen Star 06'
[Detect:Info]	Searching for tablet 'XP-Pen Star 06C'
[Detect:Info]	Searching for tablet 'XP-Pen Star G430'
[Detect:Info]	Searching for tablet 'XP-Pen Star G430S'
[Detect:Info]	Searching for tablet 'XP-Pen Star G540 Pro'
[Detect:Info]	Searching for tablet 'XP-Pen Star G540'
[Detect:Info]	Searching for tablet 'XP-Pen Star G640 V2'
[Detect:Info]	Searching for tablet 'XP-Pen Star G640'
[Detect:Info]	Searching for tablet 'XP-Pen Star G640S'
[Detect:Info]	Searching for tablet 'XP-Pen Star G960'
[Detect:Info]	Searching for tablet 'XP-Pen Star G960S Plus'
[Detect:Info]	No tablet found.

```

jamesbt365, 2021-11-10:
Help > export diagnosis instead of dumping the console, upload it as a txt file please.

Coolykoen, 2021-11-10:
Maybe i found the issue, will try changing the profile to match the detected hardware:
[image]

Coolykoen, 2021-11-10:
nvm thats not it (im dumb, scrolled a bit further down the profile and it was already in there)

Coolykoen, 2021-11-10:
Here is the dump:
[dump.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7515259/dump.txt)

jamesbt365, 2021-11-10:
can you: right click on start menu > device manager > view > device by container > look for your tablet there > open the container (arrow on the left) > screenshot device manager

Coolykoen, 2021-11-10:
Sure :)
[image]

jamesbt365, 2021-11-10:
Tablet > device string reader:
9580 in the top box 110 in the middle and click dump all, save that and send it here

Coolykoen, 2021-11-10:
Here you go, also thanks for your help ^^
[dump2.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7515338/dump2.txt)

Mrcubix, 2021-11-10:
this tablet require the use of WinUSB and pre-release build to work, but first, do these steps:

close otd, get the oem driver, install it and run it then following the next series of steps:

- https://desowin.org/usbpcap/thankyou.html?file=1.5.4.0/USBPcapSetup-1.5.4.0.exe Download and install this tool

- When done, head to C:\Program Files\USBPcap in the explorer (you can probably copy paste that path in the path bar)

- Once you're there, start USBPcapCMD.exe

- A console will appear, answer n at the first prompt and reboot your pc

- Head back to C:\Program Files\USBPcap,

- Start USBPcapCMD.exe,

- screenshot the console.

Mrcubix, 2021-11-10:
na, we are just getting the data necessary to know which interface you'll have to install WinUSB on and the config you'll have to use

Coolykoen, 2021-11-10:
i see, alright sounds good :)

Coolykoen, 2021-11-14:
Hey, as promised im back, and uh, the usbpcap console didnt give any prompt so i couldnt do this step:
"A console will appear, answer n at the first prompt and reboot your pc"
But here is the console screenshot, with the oem driver installed and running in the background.
[image]

Mrcubix, 2021-11-14:
- type 3 then enter,
- name the file. Press enter and do the following steps:

- Move for 5s
- Unplug the tablet from your computer (remove the usb cable from the port),
- Wait 2 seconds,
- Plug in your tablet again (connect your tablet to your pc ),
- Wait 2 seconds,
- Move for 5s
- Close the console (red cross),
- A new file has been generated with the name you gave in the same folder,
- Upload that file here

Coolykoen, 2021-11-14:
[tablet.pcap.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7533732/tablet.pcap.txt)
Uploaded as .txt again because .pcap is not allowed :D

Mrcubix, 2021-11-14:
- Download [Zadig](https://github.com/pbatard/libwdi/releases/download/b755/zadig-2.6.exe)
- Start Zadig
- Options > show all devices
- Screenshot the list of devices

Coolykoen, 2021-11-14:
[image]

Mrcubix, 2021-11-14:
Install WinUSB (driver on the right side) to `PenTablet (Interface 1)`

Coolykoen, 2021-11-14:
So after doing that, it said driver installation successful, but opentablet driver is not detecting it yet? 
should i restart both? maybe restart the pc?

[image]
[image]

Mrcubix, 2021-11-14:
export a diag now, a new descriptor should have appeared

Coolykoen, 2021-11-14:
in which program? do i do the usbpcap thing again?

Mrcubix, 2021-11-14:
In OTD obviously

Coolykoen, 2021-11-14:
Both of them here:

[new.json.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7533838/new.json.txt)
[new.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7533839/new.txt)

Mrcubix, 2021-11-14:
https://cdn.discordapp.com/attachments/615939905566015493/890625738691211334/OpenTabletDriver.win-x64.zip you'll need to get this build to continue,
- Close the previous release of OTD
- Extract the build to it's own folder,
- Create a folder named `userdata` at the same location as the exes
- Start the UX
- Export another diag there.

Coolykoen, 2021-11-14:
So, i cant do one of the two diags cause of this:
[image]
(yes i did replug, got an error that otd deamon crashes, i clicked yes to restart the deamon, same but still got the same error when trying again)

here's the other dia
[newnew.json.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7533854/newnew.json.txt)
g:

Mrcubix, 2021-11-14:
i asked for a diag, not a string dump, so let me check that 2s

Mrcubix, 2021-11-14:
i don't see the tablet in the diagnostic file, are you sure it's plugged in?

Coolykoen, 2021-11-14:
yup :D

Coolykoen, 2021-11-14:
since the winusb driver was installed, my mouse doesnt move when i use the tablet so maybe that didnt work out somehow?

Coolykoen, 2021-11-14:
Ill restart my pc, and do a diag again to make sure

Mrcubix, 2021-11-14:
just replug

Coolykoen, 2021-11-14:
[newnewnew.json.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7533882/newnewnew.json.txt)

Mrcubix, 2021-11-14:
[H610 Pro.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7533885/H610.Pro.zip)
extract this in `Configurations/Huion` then either go to `Tablets > Detect tablet` or Restart OTD, you tablet should be detected and your cursor should be moving

Coolykoen, 2021-11-14:
In the normal OTD release? or the custom version you sent?

Mrcubix, 2021-11-14:
Pre-release, the latest build i sent

Coolykoen, 2021-11-14:
In the OTD version you sent, it detect the tablet now (progress :D) however, the cursor wont move.

Mrcubix, 2021-11-14:
go to `Tablet > tablet debugger` move your pen around, does it report any kind of data?

Coolykoen, 2021-11-14:
nope, this windows stays empty
[image]

Mrcubix, 2021-11-14:
export a diag again and upload it here

Coolykoen, 2021-11-14:
[newnewnewnew.json.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7533912/newnewnewnew.json.txt)

Mrcubix, 2021-11-14:
wait, did you install WinUSB on all 3 interfaces without my knowledge?

Coolykoen, 2021-11-14:
i didnt though
interface 2 has the original driver
int 0 seems to have a different version of winusb (even though i used the cleaning batch file before starting)

So no idea how that got there
[image]
[image]
[image]

Coolykoen, 2021-11-14:
Will do, if you dont mind, another time. probably tomorrow. Thanks for your help so far

Coolykoen, 2021-11-15:
[image]

Mrcubix, 2021-11-15:
[image]
uninstall both of those, tick the box at the last step to remove the driver, then replug, and install WinUSB on `Pentablet interface 1` using zadig

Coolykoen, 2021-11-15:
just to make sure we're on the same page, here's the before:
[image]

Now i'll install winusb to int 1:
[image]
[image]
[image]

And here's what that looks like after installing:
[image]

Yea it looks better now, lets try OTD again...
Okay, more progress...

this time, i ...kind of... have a working pen.
It doesnt move the mouse almost at all, but sometimes when just tapping and moving quickly, i do see the cursor jump around a bit here and there.

Mrcubix, 2021-11-15:
export a diagnostic file now from pre-release

Coolykoen, 2021-11-15:
[newnewnewnewnew.json.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7540830/newnewnewnewnew.json.txt)

Mrcubix, 2021-11-15:
you indeed installed WinUSB on another interface before, anyway, use this config  
[H610 Pro 2048.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7540841/H610.Pro.2048.zip)
anyway, delete the old config and move this to `Configurations\Huion`

Coolykoen, 2021-11-15:
YES that did it!
it moves, and i think it moves perfectly at that.

however, the cursor only moves when dragging, not when hovering, so the mouse is always holding left click.
Hmm, any idea whats causing that?

Mrcubix, 2021-11-15:
lemme check the parser, i'll edit that message when done

Coolykoen, 2021-11-15:
Im dragging the cursor around on that debugger window, but nothing... just always looks like this:
[image]
I tried turning on "enable data recording" but still nothing

Mrcubix, 2021-11-15:
Which cursor move, windows ink (small dot or diamond) or windows cursor

Coolykoen, 2021-11-15:
the small dot one
also it works when otd is not even running... so windows is probably still screwing with it
[newnewnewnewnewnew.json.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7541037/newnewnewnewnewnew.json.txt)

Coolykoen, 2021-11-15:
[image]
This is looking really good now

the tablet raw data is like:
C0 is for hover, it changes to C1 when dragging

The four next ones are movement, and the last two are for pressure

the cursor now doesnt move at all (neither dot nor mouse)
But this data is very useful i imagine

Mrcubix, 2021-11-15:
So first of all, Huion is initializing their tablet rong somehow, second, what's the second byte when you press aux keys (tablet keys, not the pen)

Coolykoen, 2021-11-15:
Well to be fair we are using a config file for the 2048 version whereas this is the H610PRO(8192)
anyway i have to go now, will continue next time, sorry for all the delays, got busy work days :)
Thanks again

Mrcubix, 2021-11-15:
funny thing is that, i already know what's the max pressure this thing will throw and it's def 2048

Coolykoen, 2021-11-16:
Question, could we possibly do this in a voice call? screenshare for example would help make things go very quickly
Then we just document here what the issues were and what we did to solve them.

because i'd like to fix it, but spending a week for it seems a bit crazy just to be able to use it haha.

if not, we can ofc continue here though.

Mrcubix, 2021-11-16:
we have a [discord server](https://discord.gg/P3GYdUS) but i don't feel like going in vc, would prob still be faster

Coolykoen, 2021-11-16:
will do once we continue, thanks

[3 more replies]

closed 2021-12-27 by jamesbt365