OpenTabletDriver/OpenTabletDriver #1624: otd wont run on start up, did I do something wrong? or it is how its supposed to be

benabenir, 2021-11-11:
# Support Request

## What I've already tried

## Diagnostic Information

jamesbt365, 2021-11-11:
As you have litreally given me no information on anything, like operating system, you have delayed this a little for yourself.

For linux, just run the service.
Windows:
- Right click OpenTabletDriver.UX.Wpf.exe (should be in %localappdata%\OpenTabletDriver\bin when installer is used) -> Create shortcut
- Right click shortcut -> Properties -> Run -> Minimized
- Press Win + R -> Type shell:startup -> Press Enter
- Move the shortcut to the folder that opened

jamesbt365, 2021-11-11:
Closing due to not filled out template and the fact that it is answered.

closed 2021-11-11 by jamesbt365