OpenTabletDriver/OpenTabletDriver #1635: Amount of exported environment variables in diagnostics might be too extreme

gonX, 2021-11-11:
## Description
OTD diagnostics reports every single environment variable set on the system, which can be problematic privacy-wise, or at worst leak a users password or an API key.

Ideally we should only report the necessary environment variables, or at least make exporting of environment variables optional, since most issues can be resolved without having to see the users environment variables.

This issue can be used to track which environment variables we're definitely interested in. For now, I can think of 2:
- `DISPLAY` 
- `WAYLAND_DISPLAY`

## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System         | Any
| OpenTabletDriver Version | 35ae3e6

Sublimelime, 2021-11-11:
We should basically drop anything that we don't _directly_ check in code somewhere (afaik only the two DISPLAY ones, maybe ones related to dotnet?). If someone has an issue that might be caused by a bad env var, we can just ask them what the value of that var is.

Just dumping all env vars is definitely bad - while I disagree with the practice (for exactly this reason, random software dumping it), some people put api keys or passwords into their env vars. Plus it leaks stuff like their real name (often people set their username to their real name), etc.

jamesbt365, 2021-11-11:
if the user is administator is useful to know, however if we reveal the user we could find out peoples names, and some people don't like that. Maybe it is possible to isolate this as well?

Sublimelime, 2021-11-11:
We should already know that from the console message generated when OTD is ran as admin, yes?

jamesbt365, 2021-11-11:
As fair as i know, this doesn't appear if the user is user the 'Administrator' account.

gonX, 2021-11-11:
> if the user is administator is useful to know, however if we reveal the user we could find out peoples names, and some people don't like that. Maybe it is possible to isolate this as well?
We don't really need the specific account name, just if its an elevated account. I think that should be an error message instead.

Sublimelime, 2021-11-11:
Yeah we should just universally determine if we're running elevated and pop an error/warning in the console

jamesbt365, 2021-11-11:
careful about approaching the administrator named account, the warning doesn't pop up already so this would need to be done some other way or the warning be fixed for that.

PR #1692 (2021-12-05): Introduce less intrusive environment variables in diagnostics
Closes #1635
closed 2021-12-05 by InfinityGhost