OpenTabletDriver/OpenTabletDriver #1641: OTD does not build nor run with linux distros that upgraded to .NET 6

X9VoiD, 2021-11-13:
## Description

OpenTabletDriver does not have a target framework entry to support .NET 6, and the .NET 6 runtime does not have backwards compatibility with .NET 5.

## Possible fixes

- Both frameworks can be supported by specifying both `.net5.0` and `.net6.0` in `TargetFrameworks`
- Just merge #1441 
- Or, embed the runtime and enable assembly trimming for smaller binary file sizes.

chent7, 2021-11-13:
For now use 5.0 binaries from the AUR for anyone looking for a temporary solution.

gonX, 2021-11-13:
> For now use 5.0 binaries from the AUR for anyone looking for a temporary solution.

AUR builds from source, so even 0.5.3.3 will be broken. Current workarounds for that version is to install `dotnet-runtime-5.0-bin` or wait for .NET6 support for OTD to be backported.

opentabletdriver-git on AUR does patch OTD with #1441, so that works for now.

chent7, 2021-11-13:
> AUR builds from source, so even 0.5.3.3 will be broken. Current workarounds for that version is to install `dotnet-runtime-5.0-bin` or wait for .NET6 support for OTD to be backported. [...]

This is what I meant.
You will also need `dotnet-sdk-5.0-bin` `dotnet-targeting-pack-5.0-bin`

Fxsch, 2021-11-14:
Trying to install those packages always ends with an error while downloading from the microsoft servers

chent7, 2021-11-14:
> Trying to install those packages always ends with an error while downloading from the microsoft servers

I did my fresh install yesterday and confirmed otd will compile. You should check the source url for the dotnet binaries from the aur.

Fxsch, 2021-11-14:
The dotnet 5 packages only failed to download when using pamac. Installing them with yay worked. Building OpenTabletDriver is failing tho. I tried both packages in the AUR and downloading from github and running the build.sh script. The errors said something about dotnet.

chent7, 2021-11-14:
> The dotnet 5 packages only failed to download when using pamac. Installing them with yay worked. Building OpenTabletDriver is failing tho. I tried both packages in the AUR and downloading from github and running the build.sh script. The errors said something about dotnet.

Any screenshots or logs would be nice. Also make sure you uninstall dotnet 6 packages just to remove some variables.

Fxsch, 2021-11-14:
https://pastebin.com/7mNm4GsQ

I also uninstalled the dotnet 6 packages

chent7, 2021-11-14:
> https://pastebin.com/7mNm4GsQ
> 
> I also uninstalled the dotnet 6 packages

Try building the non-git version. Git version should compile with dotnet 6.

Fxsch, 2021-11-14:
That worked

PR #1441 (2021-11-24): Add support for .NET 6.0
and use Directory.Build.props to deduplicate some properties.

- Fixes #1641
closed 2021-11-24 by InfinityGhost