OpenTabletDriver/OpenTabletDriver #1649: Outdated wiki should be taken down

X9VoiD, 2021-11-16:
## Description

People can't see that the wiki here is already outdated because it's at the top, and users that enters the wiki through a topic link (errors from OTD for example) will not be able to see it.

The only solution would be to remove all content and only keep the link to the updated wiki for each legacy pages.

gonX, 2021-12-01:
Setting to high priority because its actively confusing users and annoying helpers.

It's better we have no info than outdated info, so everything should be put on the [new wiki](https://github.com/OpenTabletDriver/OpenTabletDriver.Web) instead.

gonX, 2021-12-06:
Fixed. Github wiki has redirects set up now

closed 2021-12-06 by gonX