OpenTabletDriver/OpenTabletDriver #1650: Add support for XP-PEN Deco FUN

Dfaker-HK, 2021-11-16:
# Tablet Information

| Name            | Value |
| --------------- | ----- |
| Manufacturer    | XP-PEN
| Product Name    |  DECO FUN
| Width           | 6.3＂
| Height          | 4＂
| Pressure Levels | 8192
| LPI             | 5080

## Diagnostic Information

[deco diagnostic.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7546113/deco.diagnostic.txt)

Mrcubix, 2021-11-16:
get the archive https://cdn.discordapp.com/attachments/628651971267788800/910145351191756820/CT_640.zip
extract it's content to `Configurations\XP-Pen` then in OTD: `Tablet > Detect Tablet` or Restart

Mrcubix, 2021-12-16:
closing as stale and merged + available in OTD.Backport

closed 2021-12-16 by Mrcubix