OpenTabletDriver/OpenTabletDriver #1651: Support Request: Wacom CTL-4100WL/K not detected with Bluetooth or USB

minhduc66532, 2021-11-16:
# Support Request

OTD keeps showing "No tablets are detected" when connecting the tablet using Bluetooth or USB
## What I've already tried

- Saw this [issue](https://github.com/OpenTabletDriver/OpenTabletDriver/issues/1293) talking about udev rules but the rule file is already in the correct folder after installing OTD from AUR
- Restart service, replug the tablet, reboot, still the same issue

## Diagnostic Information
[dunp.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7546659/dunp.txt)

SnowLire, 2021-11-16:
I assume the diagnostic was made with the tablet in BT mode, in which case the VID and PID of the device is different than what is in the configuration.
As for the USB issue, it's a known issue with all CTL-4/6100 and [here are the steps](https://github.com/linuxwacom/xf86-input-wacom/wiki/Known-Issues#issue-my-wacom-intuos-ctl-4100-or-ctl-6100-seems-to-be-miscalibrated-and-has-a-dimmed-status-led) to resolve it.

minhduc66532, 2021-11-18:
@Mrcubix @SnowLire Sorry for the late reply but there is a new problem (also the dump above was made when the tablet was connecting to the laptop using USB NOT Bluetooth. The steps from the guide also did not solve the issue). The daemon failed to start for some reason with this error log:
```
### Searching for tablets stuff with no warning about wacom driver anymore
[Detect:Info]   No tablet found.
Unhandled exception. System.Net.Internals.SocketExceptionFactory+ExtendedSocketException (111): Connection refused /run/user/1000
   at System.Net.Sockets.Socket.DoConnect(EndPoint endPointSnapshot, SocketAddress socketAddress)
   at System.Net.Sockets.Socket.Connect(EndPoint remoteEP)
   at WaylandNET.Client.WaylandClientConnection.ConnectToDisplay(String display)
   at WaylandNET.Client.WaylandClientConnection..ctor(String display)
   at OpenTabletDriver.Desktop.Interop.Display.WaylandDisplay..ctor() in /home/minhduc/.cache/yay/opentabletdriver/src/OpenTabletDriver-0.5.3.3/OpenTabletDriver.Desktop/Interop/Display/WaylandDisplay.cs:line 18
   at OpenTabletDriver.Desktop.Interop.SystemInterop.ConstructLinuxDisplay() in /home/minhduc/.cache/yay/opentabletdriver/src/OpenTabletDriver-0.5.3.3/OpenTabletDriver.Desktop/Interop/SystemInterop.cs:line 104
   at OpenTabletDriver.Desktop.Interop.SystemInterop.<>c.<.cctor>b__18_3() in /home/minhduc/.cache/yay/opentabletdriver/src/OpenTabletDriver-0.5.3.3/OpenTabletDriver.Desktop/Interop/SystemInterop.cs:line 96
   at System.Lazy`1.ViaFactory(LazyThreadSafetyMode mode)
   at System.Lazy`1.ExecutionAndPublication(LazyHelper executionAndPublication, Boolean useDefaultConstructor)
   at System.Lazy`1.CreateValue()
   at System.Lazy`1.get_Value()
   at OpenTabletDriver.Desktop.Interop.SystemInterop.get_VirtualScreen() in /home/minhduc/.cache/yay/opentabletdriver/src/OpenTabletDriver-0.5.3.3/OpenTabletDriver.Desktop/Interop/SystemInterop.cs:line 60
   at OpenTabletDriver.Desktop.Settings.get_Default() in /home/minhduc/.cache/yay/opentabletdriver/src/OpenTabletDriver-0.5.3.3/OpenTabletDriver.Desktop/Settings.cs:line 331
   at OpenTabletDriver.Daemon.DriverDaemon.ResetSettings() in /home/minhduc/.cache/yay/opentabletdriver/src/OpenTabletDriver-0.5.3.3/OpenTabletDriver.Daemon/DriverDaemon.cs:line 228
   at OpenTabletDriver.Daemon.DriverDaemon.LoadUserSettings() in /home/minhduc/.cache/yay/opentabletdriver/src/OpenTabletDriver-0.5.3.3/OpenTabletDriver.Daemon/DriverDaemon.cs:line 89
   at System.Threading.Tasks.Task.<>c.<ThrowAsync>b__140_1(Object state)
   at System.Threading.QueueUserWorkItemCallbackDefaultContext.Execute()
   at System.Threading.ThreadPoolWorkQueue.Dispatch()
   at System.Threading._ThreadPoolWaitCallback.PerformWaitCallback()
/usr/bin/otd: line 13: 11522 Aborted                 (core dumped) dotnet ./OpenTabletDriver.Daemon.dll -c ./Configu
```
I followed the FAQ and updated the new udev rule, here is the content of `/etc/udev/rules.d/99-opentabletdriver.rules`: [99-opentabletdriver.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7559837/99-opentabletdriver.txt)

Mrcubix, 2021-11-18:
Export a diag

minhduc66532, 2021-11-21:
OTD crashed every time I tried to do so

Mrcubix, 2021-12-16:
[99-opentabletdriver.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7730988/99-opentabletdriver.zip)
extract the content of that zip file with the following path `sudo [location] /etc/udev/rules.d/99-opentabletdriver.rules` (replace `[location]` with the file location).
Once done, do:
`sudo udevadm control --reload-rules`,
and for good measure, reload the daemon using:
`systemctl --user restart opentabletdriver.service`.

Once done, start the ux and export a diagnostic file (possibly without the config in the `Configurations` folder)

minhduc66532, 2021-12-20:
I actually reinstalled arch and everything is working correctly now, thank you for the solution, and sorry for not closing the issue sooner.

closed 2021-12-20 by minhduc66532