OpenTabletDriver/OpenTabletDriver #1654: Support Request:

ICHINOYA, 2021-11-17:
# Support Request
"OTD cannot connect to Internet. Only local plugins will be shown."
[image]

## What I've already tried
UAC Enabled
PC Rebooted
Network Adapter Restarted

## Diagnostic Information
{
  "App Version": "OpenTabletDriver v0.5.3.3",
  "Operating System": {
    "Platform": 2,
    "ServicePack": "",
    "Version": "10.0.19042.0",
    "VersionString": "Microsoft Windows NT 10.0.19042.0"
  },
  "Environment Variables": {
    "PROCESSOR_LEVEL": "6",
    "GTK_BASEPATH": "C:\\Program Files (x86)\\GtkSharp\\2.12\\",
    "COMPUTERNAME": "ICHINOYAPC",
    "RG_GPU_FRAMEWORK_ENGINE_RESOURCEDIR": "C:\\Program Files\\Red Giant\\RGFX",
    "PATHEXT": ".COM;.EXE;.BAT;.CMD;.VBS;.VBE;.JS;.JSE;.WSF;.WSH;.MSC",
    "PROCESSOR_ARCHITECTURE": "AMD64",
    "CommonProgramFiles": "C:\\Program Files\\Common Files",
    "CommonProgramFiles(x86)": "C:\\Program Files (x86)\\Common Files",
    "USERPROFILE": "C:\\Users\\ICHINOYA",
    "HOMEPATH": "\\Users\\ICHINOYA",
    "PROCESSOR_REVISION": "2d07",
    "INTEL_DEV_REDIST": "C:\\Program Files (x86)\\Common Files\\Intel\\Shared Libraries\\",
    "MIC_LD_LIBRARY_PATH": "C:\\Program Files (x86)\\Common Files\\Intel\\Shared Libraries\\compiler\\lib\\mic",
    "DriverData": "C:\\Windows\\System32\\Drivers\\DriverData",
    "PSModulePath": "C:\\Program Files\\WindowsPowerShell\\Modules;C:\\Windows\\system32\\WindowsPowerShell\\v1.0\\Modules",
    "USERNAME": "ICHINOYA",
    "SystemRoot": "C:\\Windows",
    "OneDrive": "C:\\Users\\ICHINOYA\\OneDrive",
    "SystemDrive": "C:",
    "TEMP": "C:\\Users\\ICHINOYA\\AppData\\Local\\Temp",
    "CommonProgramW6432": "C:\\Program Files\\Common Files",
    "APPDATA": "C:\\Users\\ICHINOYA\\AppData\\Roaming",
    "TMP": "C:\\Users\\ICHINOYA\\AppData\\Local\\Temp",
    "HOMEDRIVE": "C:",
    "PUBLIC": "C:\\Users\\Public",
    "ProgramData": "C:\\ProgramData",
    "SESSIONNAME": "Console",
    "USERDOMAIN": "ICHINOYAPC",
    "OS": "Windows_NT",
    "LOCALAPPDATA": "C:\\Users\\ICHINOYA\\AppData\\Local",
    "windir": "C:\\Windows",
    "PROCESSOR_IDENTIFIER": "Intel64 Family 6 Model 45 Stepping 7, GenuineIntel",
    "ComSpec": "C:\\Windows\\system32\\cmd.exe",
    "ALLUSERSPROFILE": "C:\\ProgramData",
    "Path": "C:\\Program Files (x86)\\Common Files\\Oracle\\Java\\javapath;C:\\Program Files\\WinRAR;C:\\Program Files (x86)\\NVIDIA Corporation\\PhysX\\Common;C:\\Program Files (x86)\\Common Files\\Intel\\Shared Libraries\\redist\\intel64\\compiler;C:\\Windows\\system32;C:\\Windows;C:\\Windows\\System32\\Wbem;C:\\Windows\\System32\\WindowsPowerShell\\v1.0\\;C:\\Windows\\System32\\OpenSSH\\;C:\\Program Files\\dotnet\\;C:\\Program Files\\Microsoft SQL Server\\Client SDK\\ODBC\\170\\Tools\\Binn\\;C:\\Users\\ICHINOYA\\AppData\\Local\\Microsoft\\WindowsApps;C:\\Users\\ICHINOYA\\.dotnet\\tools;C:\\adb;c:\\Program Files (x86)\\Microsoft SQL Server\\100\\Tools\\Binn\\;c:\\Program Files\\Microsoft SQL Server\\100\\Tools\\Binn\\;c:\\Program Files\\Microsoft SQL Server\\100\\DTS\\Binn\\;C:\\Program Files (x86)\\Microsoft SQL Server\\150\\DTS\\Binn\\;C:\\Program Files (x86)\\GtkSharp\\2.12\\bin;C:\\Users\\ICHINOYA\\AppData\\Local\\Programs\\Python\\Python310\\Scripts\\;C:\\Users\\ICHINOYA\\AppData\\Local\\Programs\\Python\\Python310\\;C:\\Users\\ICHINOYA\\AppData\\Local\\Microsoft\\WindowsApps;C:\\Users\\ICHINOYA\\.dotnet\\tools;C:\\Program Files\\Azure Data Studio\\bin;C:\\Users\\ICHINOYA\\AppData\\Local\\GitHubDesktop\\bin",
    "ProgramFiles(x86)": "C:\\Program Files (x86)",
    "NUMBER_OF_PROCESSORS": "12",
    "ProgramFiles": "C:\\Program Files",
    "USERDOMAIN_ROAMINGPROFILE": "ICHINOYAPC",
    "LOGONSERVER": "\\\\ICHINOYAPC",
    "ProgramW6432": "C:\\Program Files"
  },
  "HID Devices": [
    {
      "DevicePath": "\\\\?\\hid#vid_258a&pid_0016&mi_01&col02#8&17ef520f&0&0001#{4d1e55b2-f16f-11cf-88cb-001111000030}",
      "VendorID": 9610,
      "ProductID": 22,
      "CanOpen": true,
      "ReleaseNumberBcd": 1,
      "ReleaseNumber": "0.1",
      "ProductVersion": 1,
[... 535 lines cut ...]
      "Time": "2021-11-17T22:31:48.4508138+03:00",
      "Group": "Display",
      "Message": "DPI Awareness enabled",
      "Level": 0
    },
    {
      "Time": "2021-11-17T22:31:48.4508389+03:00",
      "Group": "IPC",
      "Message": "Disconnected from a client.",
      "Level": 0
    },
    {
      "Time": "2021-11-17T22:31:48.4508645+03:00",
      "Group": "IPC",
      "Message": "Connected to a client.",
      "Level": 0
    },
    {
      "Time": "2021-11-17T22:31:48.450889+03:00",
      "Group": "Display",
      "Message": "DPI Awareness enabled",
      "Level": 0
    }
  ]
}

jamesbt365, 2021-11-17:
attach the diag as a txt please, don't just throw it in.

ICHINOYA, 2021-11-17:
> attach the diag as a txt please, don't just throw it in.

Yeap

jamesbt365, 2021-12-31:
Usually caused by a rate limit to github or something to do with your firewall.

closed 2021-12-31 by jamesbt365