OpenTabletDriver/OpenTabletDriver #1655: Support Request: installed dotnet with spoon (package manager), did not run.

brickyosu, 2021-11-18:
# Support Request
i installed dotnet with spoon bc i wanna use my wacom tablet for school except i dont have admin on this school laptop.
i runned opentabletdriver wpf and it still prompts to install dotnet

## What I've already tried
installing with different package manager like chocolatey, did not work either.

## Diagnostic Information
nope, i cannot.

jamesbt365, 2021-11-18:
This is probably due to you not having admin rights.

https://cdn.discordapp.com/attachments/843628282506379295/910552649798156339/win-x64.7z.001

This build of otd has dotnet embedded within it. Just extract it and go for it

brickyosu, 2021-11-18:
> This is probably due to you not having admin rights.
> 
> https://cdn.discordapp.com/attachments/843628282506379295/910552649798156339/win-x64.7z.001
> 
> This build of otd has dotnet embedded within it. Just extract it and go for it

ok ill try, ill keep updating

brickyosu, 2021-11-18:
> This is probably due to you not having admin rights.
> 
> https://cdn.discordapp.com/attachments/843628282506379295/910552649798156339/win-x64.7z.001
> 
> This build of otd has dotnet embedded within it. Just extract it and go for it

Thanks! It worked.

jamesbt365, 2021-11-18:
Closing as resolved.

closed 2021-11-18 by jamesbt365

brickyosu, 2021-11-18:
sadly i lost my wacom tablet pen, im trying my best to find it