OpenTabletDriver/OpenTabletDriver #1660: Monitor mapping broken on Gnome 4.41

X9VoiD, 2021-11-21:
## Description

OTD monitor mapping is broken on Gnome 4.41 and is constrained to a single display.

InfinityGhost, 2021-11-29:
Is this Wayland or Xorg? I'm unable to test this as I've only got a single display setup.

X9VoiD, 2021-11-29:
Both.

gonX, 2021-12-01:
Possibly duplicate of #1137?

gonX, 2021-12-20:
According to the reporter in #1137 this seems to no longer be the case. Can you confirm @X9VoiD ?

gonX, 2021-12-25:
I've tested this on GNOME 41.2 and I am unable to reproduce

gonX, 2021-12-26:
Reopening since someone on 41.2 seems to have the issue

[discord convo link](https://discord.com/channels/615607687467761684/723399565780189234/924617534949441556)

gonX, 2021-12-26:
Potential leads: artist mode and left (mapped) monitor mapped below right monitor

sburris0, 2023-07-07:
I believe I'm experiencing this on GNOME 44.2/Wayland/nixos.

From what I can tell, this bug is centered around configured display positions. I don't notice a difference between absolute and artist mode with my Huion Kamvas 16 2021. Pen position looks correct in tablet debugger view.

I've played with monitor positions a lot, and the cursor seems to be:
* too high or too low, depending on monitor positions
* always too far to the left, varying amounts

Note that tablet is monitor 2 in the below images

## Cases of incorrect position
Cursor too high and to the left:
[image]

Cursor WAY too high and to the left:
[image]

Cursor too low and to the left:
[image]

Cursor too low and WAY too far to the left:
[image]

Cursor too high and to the left:
[image]

Cursor very high and way too far to the left:
[image]

## Cases of correct position
If I re-enable my third display, I can actually get the correct cursor positions with these configurations:
[image]
[image]
[image]
[image]

But no others that I have found.

This seems pretty reliable on my setup, so I'd be happy to run experiments or provide any other info

jamesbt365, 2025-01-24:
Not really seeing any complaints about mapping not working on gnome anymore? I assume this is probably fixed upstream already?

gonX, 2025-01-24:
> Not really seeing any complaints about mapping not working on gnome anymore? I assume this is probably fixed upstream already?

Yeah, going to assume the same. I've not heard many GNOME complaints the last couple of months.

Closing as assumed fixed upstream.

closed 2025-01-24 by gonX