OpenTabletDriver/OpenTabletDriver #1661: Support Request: Auxiliary buttons not working on Deco 01 v2

8ryo, 2021-11-22:
As title says, the auxiliary buttons aren't working on the XP-Pen Deco 01 v2.

[image]

All 8 buttons are reporting 1 through to 128 in the tablet debugger

I've tried different output modes (e.g. Absolute, windows ink absolute etc. ), nothing has worked. Also restarted my pc to see if it would work, to no avail. 

Tried it with all plugins disabled, still doesn't work.

Seems like a configuration issue but someone else would have brought up this bug by now so I'm gonna presume there's something wrong on my end.

Here's my config:

[image]

[image]

I have script runner and OEM kill enabled

jamesbt365, 2021-11-22:
Please export a diagnostic for me. Help > export diagnostics, save that file and attach it here as a txt file.

8ryo, 2021-11-22:
[Deco 01 v2.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7584830/Deco.01.v2.txt)

mb it probably said to do that earlier, here you go

Mrcubix, 2021-11-22:
let's do something simple, go to `Tablet > Tablet debugger`, there press one of your button, if on the right it say aux and then show all 8 buttons then the tablet send the data as intended, if not then a parser modification is required

8ryo, 2021-11-22:
> * if on the right side, it say aux and then show all 8 keys then the tablet send the data as intended,

You can see it in the top screenshot, it doesn't say aux, it shows 1,2,4,8,16... for all buttons 1-8 respectively in the position part on the tablet report.

[image]

This is me pressing the very top button on the tablet

8ryo, 2021-11-22:
It seems to be reporting back button presses as positions on the tablet as far as I can tell

8ryo, 2021-11-23:
Raw tablet data from each button press, 0 degrees orientation, going from top to bottom:

02 F0 01
02 F0 02
02 F0 04
02 F0 08
02 F0 10
02 F0 20
02 F0 40
02 F0 80

8ryo, 2021-11-23:
And then tablet report shows for each button press respectively:

Position: [1,0]
Position: [2,0]
Position: [4,0]
Position: [8,0]
Position: [16,0]
Position: [32,0]
Position: [64,0]
Position: [128,0]

ReportID: 0 
Pressure: 0
PenButtons: [False, False]

These last 3 remain for all button presses

gonX, 2021-11-23:
I gave this a quick look, some quick notes:

- This is fixed on master, so if using a pre-release version is fine with you (note that plugin support is spotty), that is an option.
- For any helpers reading: the XP Pen parser in 0.5.3 does not support aux reports from this tablet, so simply changing the parser won't fix it. A parser backport of some sort is needed - might be possible with Cubix' Backports, but not immediately from what I can see.

Alternatively you can wait for OTD 0.6 to come out, though there is no release date set yet.

Mrcubix, 2021-11-23:
what does the first and second bytes become when dragging and when out of range?

Mrcubix, 2021-11-23:
[OTD.Backport.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7588177/OTD.Backport.zip)

- drag and drop this in `Plugin > Plugin manager`, then head to the tool tab > OTD.Backport > tick enable, then save
- `Tablets > Detect tablet` and now buttons should work

8ryo, 2021-11-23:
I installed the plugin, didn't work on 0.4.1.0 or 0.5.3.3

jamesbt365, 2021-11-23:
Make sure you do it on 0.5.3.3 and export a diag with help > export diagnostics if it doesn't work

Mrcubix, 2021-11-23:
this plugin doesn't touch the tablet at all, only configurations , and parsers from those who need it

8ryo, 2021-11-23:
Ok kinda weird, I managed to reproduce the issue 2-3 times before but now I can't reproduce it for some reason. Either way, the buttons still don't work unfortunately

Mrcubix, 2021-11-23:
go to the tablet debugger and press them, check if they appear on the right side

8ryo, 2021-11-23:
Yeah there is a response but they appear as positions like before

Position: [1,0]
Position: [2,0]
Position: [4,0]
Position: [8,0]
Position: [16,0]
Position: [32,0]
Position: [64,0]
Position: [128,0]

Mrcubix, 2021-11-23:
that's abnormal for sure, the second byte is F0 and has the 5 byte set, which should press the button, Check the second byte when dragging and when out of range

8ryo, 2021-11-23:
Hover:

[image]

Drag:

[image]

Out of range: 

[image]

Bottom pen button will show A2/A3 for hover/drag respectively

Top pen button will show A4/A5 for hover/drag respectively

8ryo, 2021-11-23:
And then for the tablet buttons it's the same as before: 
02 F0 01
02 F0 02
02 F0 04
02 F0 08
02 F0 10
02 F0 20
02 F0 40
02 F0 80

Mrcubix, 2021-11-23:
send the the tablet config you'll find in `Configurations\XP-Pen\Deco 01 V2.json`

8ryo, 2021-11-23:
[Deco 01 V2.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7590887/Deco.01.V2.txt)

Mrcubix, 2021-11-23:
- tool > OTD.BackPort > tick enable > tick force apply > save > untick force apply 
- `Tablets > Detect tablet`

8ryo, 2021-11-23:
My tablet has stopped working

Mrcubix, 2021-11-23:
export a diagnostic file

8ryo, 2021-11-23:
[diag.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7590923/diag.txt)

Mrcubix, 2021-11-23:
- first, i told you to untick force apply in the tool,
- second, nothing in the diag indicate that it stop working in any way
- third, `Tablet > Tablet debugger`, press aux keys, see if they appear on the right, also move your pen, a blue dot should appear in the visualizer

8ryo, 2021-11-23:
Ok can you check this diag to see if I'm doing anything wrong this time:
[diag2.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7590940/diag2.txt)

This is my screen now:
[image]

Mrcubix, 2021-11-23:
`Tablet > device string reader` > type the following:

- 10429
- 2309

then presse dump all and upload that here

8ryo, 2021-11-23:
[stringdump.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7590987/stringdump.txt)

Mrcubix, 2021-11-23:
[OTD.Backport.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7591002/OTD.Backport.zip)

- uninstall OTD.Backport via the plugin manager, 
- install this new version
- Go to Tool > OTD.Backport > tick enable and force apply > save > untick force apply
- `Tablet > Detect tablet`

8ryo, 2021-11-23:
The cursor is stuck in the top left corner of the screen for both hover/drag
[diag.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7591053/diag.txt)
.

Mrcubix, 2021-11-23:
[Deco 01 V2.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7591062/Deco.01.V2.zip)
place the json file inside of the archive in `Configurations\XP-Pen` replace the old one when asked, then re-detect

8ryo, 2021-11-23:
ok pen works but only bottom half of tablet area is detected, tablet buttons sends the cursor down to bottom left
[diag.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7591136/diag.txt)

this is when i hover about 2/3 from the top of the drawing area
[image]

Mrcubix, 2021-11-23:
try to get maximum position by moving to the bottom right of the tablet

8ryo, 2021-11-23:
yeah i can get there no issue, its aligned properly there

Mrcubix, 2021-11-23:
i should had added that i need the value

8ryo, 2021-11-23:
02 A0 00 00 ...

Mrcubix, 2021-11-23:
i do not need raw value, i need the value on the right where it say Position

8ryo, 2021-11-23:
[0, 0]

8ryo, 2021-11-23:
oh mb i have it left handed

8ryo, 2021-11-23:
[50791, 31750] that's as far as i can get it

8ryo, 2021-11-23:
Do you have the old json, it might have been area refresh issue, the driver sometimes messes up the drawing area

Mrcubix, 2021-11-23:
[Deco 01 V2.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7591206/Deco.01.V2.zip)
use this config, after re-plugging your tablet ,you should be able to use it entirely

8ryo, 2021-11-23:
yeah the full area works now

Mrcubix, 2021-11-23:
aux keys shoud also work, check that

8ryo, 2021-11-23:
no unfortunately, still sends the cursor down to the bottom left

8ryo, 2021-11-23:
but there is a response which is a good sign I suppose

Mrcubix, 2021-11-23:
you mean top left, but anyway

`Tablet > Tablet debugger` and i assume that when you press an aux key, no aux key array appear on the right?

which is weird because when you press an aux key, you state it's `F0` which convert to `11110000` which mean the bit at index 4 (5th bit) is set (= 1)

8ryo, 2021-11-23:
[image]

This is when I press a tablet button

Mrcubix, 2021-11-23:
send the deco 01 config you have rn

8ryo, 2021-11-23:
[Deco 01 V2.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7591276/Deco.01.V2.txt)

[diag.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7591280/diag.txt)

Mrcubix, 2021-11-23:
[OTD.Backport-dll.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7591290/OTD.Backport-dll.zip)

close OTD, head to `%localappdata%/opentabletdriver/Plugins/OTD.Backport` and extract the zip file there, replace the dll when ask, start OTD again and everything should work

8ryo, 2021-11-23:
Works initially but when button is pressed, tablet is no longer detected

Mrcubix, 2021-11-23:
export a diag

8ryo, 2021-11-23:
[diag.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7591319/diag.txt)

Mrcubix, 2021-11-23:
i need a diag after the issue actually happen, don't restart OTD

8ryo, 2021-11-23:
[diag.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7591325/diag.txt)

Mrcubix, 2021-11-23:
this is pressing aux keys right?

8ryo, 2021-11-23:
yep

Mrcubix, 2021-11-23:
[OTD.Backport-dll.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7591386/OTD.Backport-dll.zip)

close OTD, replace the DLL in `%localappdata%/opentabletdriver/Plugins/OTD.Backport`

start OTD again, press the any aux key, export a diag

[37 more replies]

closed 2021-12-20 by Mrcubix