OpenTabletDriver/OpenTabletDriver #1664: absolute position for touch

UltimateBoi, 2021-11-25:
## Description

Is there anything that lets me use absolute position with touch on my touchpad? right now it only works on the pen.

Mrcubix, 2021-11-25:
State what tablet you're using, but i'm pretty sure this has already been added in pre-release.

UltimateBoi, 2021-11-25:
CTH-670

InfinityGhost, 2021-12-05:
> State what tablet you're using, but i'm pretty sure this has already been added in pre-release and master.

Depends on what they're actually looking for. This sounds like they want touch input to be interpreted as position data in which to move the cursor.

UltimateBoi, 2021-12-10:
> > State what tablet you're using, but i'm pretty sure this has already been added in pre-release and master.
> 
> Depends on what they're actually looking for. This sounds like they want touch input to be interpreted as position data in which to move the cursor.

Basically it clicks on the spot on the screen without me having to actually drag my finger across the pad it just clicks on the position when I touch. Its already getting this data but its not implemented yet. Just like absolute position for the pen but for touch basically.

Espionage724, 2022-06-12:
I would like to see this! I would like to use my finger, instead of a stylus to have the mouse move in absolute mode. If I tap the bottom-right of my tablet, the cursor should jump there.

gonX, 2022-06-12:
The feature is planned, its somewhat on the backlog but we are of course taking PR's

Mrcubix, 2024-07-21:
Update: Touch support is available via the `Enhanced Output Modes` plugin in Plugins > Plugin Manager, comes with customizable settings for each tablets in the `Filters` tab under `Touch Settings`. 

On top of adding support for touch, I've also taken a large amount of time & rewrites to implement non-native touch gestures,
results may vary from one person to another, but if you want them, try them & open a issue if you find anything weird or straight up bad. Check out `Touch Gesture` in the Plugin Manager.

Both of those have separate versions for 0.5.x & 0.6.x .

I'm still learning how to do UX & how to properly make that backend part so expect some issues, hopefully nothing critical.

Touch support in itself could be made available OOB in OTD, but that would require a lot of config & some UX changes (And Eto.Forms makes me crazy).

jamesbt365, 2024-09-12:
Click "show plugin wiki" in the plugin manager.

Mrcubix, 2024-09-12:
I primarily use it on Linux, on a raspberry Pi 4 connected & wirelessly.