OpenTabletDriver/OpenTabletDriver #1666: Similar tablets are not separately addressible

X9VoiD, 2021-11-26:
## Description

OTD (#1098) internally uses a non-unique [`DeviceIdentifier`](https://github.com/OpenTabletDriver/OpenTabletDriver/blob/master/OpenTabletDriver.Plugin/Tablet/DeviceIdentifier.cs) to identify currently connected tablets. These create some issues.

- `[Core]`: It's impossible to separately manipulate these similar devices.
- `[Desktop Impl]`: A single `Settings` apply for all of them.

gonX, 2022-11-07:
Would adding a randomly generated ID to the DeviceIdentifier (which btw was moved to [`OpenTabletDriver/Tablet/DeviceIdentifier.cs`](https://github.com/OpenTabletDriver/OpenTabletDriver/blob/master/OpenTabletDriver/Tablet/DeviceIdentifier.cs)) help? I'm not sorting through a ~4000 lines changed PR to get a clue ;)

X9VoiD, 2022-11-07:
Doesn't help, current detection scheme is a lot worse now (due to the inputdevicetree 0.6.x refactoring) and only allows one "tablet" per configuration.

Currently it does this
```
foreach config
    find first matching device
    find aux endpoints
    add new device tree
```

So it's impossible to get multiple device of the same configuration to be "detected".

fredemmott, 2022-12-25:
On the topic of unique identifiers, copying from discord for reference:

VID+PID + (count-of-that-vidpid-when-OTD-starts-or-tablet-plugged-in) would probably 'good enough' for most when a serial number isn't available

e.g. i I had two Intuos S + and an H640P:
- 0x056a037601: intuos s 1
- 0x256c006d01 : h640p 1 (counter is 1 as it's a different model)
- 0x056a037602: intuos s 2

If the order's wrong, replug.