OpenTabletDriver/OpenTabletDriver #1667: Similar tablets share a single instance of `PipelineManager`

X9VoiD, 2021-11-26:
## Description

The bug will be felt with `IPipelineElement` instances that keep state.

An example is button tracking. If the same button is pressed for more than one tablet, only one button press will be generated, and only one button release will also be received. On that same note, bindings that implement some sort of `toggle` will carry over the toggle state to other devices. `WindowsInk`'s toggle eraser binding will have erroneous behaviour in such cases and can only be fixed in OTD.

The worse example would be broken relative mode when more than one of the similar tablets are generating input. On a correct implementation, they will move relatively even if they're both sending input.

Let's setup a simple case, both pens in tablet range.

- For pen 1, move left
- Then pen 2, move right

Expected behaviour:

- Cursor will go left and then right, with slight vibrations in cursor.

Actual behaviour:

- Since they share a single `PipelineManager`, the delta will be taken from wherever data comes first. We're lucky if it's the same tablet, but in case that it's not, the cursor will jump.
- Fun take: Cursor will become `Schrodinger's Cursor`. It will jump between two different places and/or act normally, randomly.

Gomesz785, 2021-12-30:
> > Plug two mouse, move one to the left, move one to the right. The cursor won't jump randomly. Move one to the right, and the other upward at the same time, cursor will move in an upright direction.

Sounds like a magic book, eg: *Get on a ladder, remove the ladder, you will fall. Now add this bottle of syrup to your hands and remove the ladder you will float*

X9VoiD, 2023-01-09:
Resolved by #2583

closed 2023-01-09 by X9VoiD