OpenTabletDriver/OpenTabletDriver #1670: application crashes when touch is enabled

Doksha, 2021-11-27:
## Description

I usually use touch mode to play osu (and is a central part to my playstyle), but for some reason it has started instantly crashing crashing when I load the preset with touch (or just enable touch), I have not changed anything to do with drivers since I last used this.
## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System         | Windows 10
| OpenTabletDriver Version | 0.5.3.3
| Tablet                   | A15

Doksha, 2021-11-27:
[diagnostics](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7612301/info.zip)
Not able to gen any diagnostics with touch enabled though since it immediately crashes

Mrcubix, 2021-11-27:
> [diagnostics](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7612301/info.zip) Not able to get any diagnostics with touch enabled though since it immediately crashes

you should be able to find `daemon.log` in `%localappdata%/opentabletdriver`, if not, you could also try starting the daemon from the folder containing it in cmd and see if there are any exception throw there you can upload in a text file here

Doksha, 2021-11-27:
[daemon.log](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7612361/daemon.txt)

Mrcubix, 2021-11-27:
you'll probably need to get the log from cmd or in Event viewer > Windows logs > Application > find .NET Runtime error related to OTD from today > post the content of the box in the middle containing `Application: OpenTabletDriver.Daemon.exe ...`

Doksha, 2021-11-27:
Application: OpenTabletDriver.UX.Wpf.exe
CoreCLR Version: 5.0.921.35908
.NET Version: 5.0.9
Description: The process was terminated due to an unhandled exception.
Exception Info: StreamJsonRpc.ConnectionLostException: The JSON-RPC connection with the remote party was lost before the request could complete.
   at StreamJsonRpc.JsonRpc.InvokeCoreAsync(JsonRpcRequest request, Type expectedResultType, CancellationToken cancellationToken)
   at StreamJsonRpc.JsonRpc.InvokeCoreAsync[TResult](RequestId id, String targetName, IReadOnlyList`1 arguments, IReadOnlyList`1 positionalArgumentDeclaredTypes, IReadOnlyDictionary`2 namedArgumentDeclaredTypes, CancellationToken cancellationToken, Boolean isParameterObject)
   at OpenTabletDriver.UX.MainForm.LoadSettingsDialog()
   at OpenTabletDriver.UX.MainForm.<ConstructMenu>b__11_3(Object sender, EventArgs e)
   at System.Threading.Tasks.Task.<>c.<ThrowAsync>b__140_0(Object state)
   at System.Windows.Threading.ExceptionWrapper.InternalRealCall(Delegate callback, Object args, Int32 numArgs)
   at System.Windows.Threading.ExceptionWrapper.TryCatchWhen(Object source, Delegate callback, Object args, Int32 numArgs, Delegate catchHandler)
   at System.Windows.Threading.DispatcherOperation.InvokeImpl()
   at MS.Internal.CulturePreservingExecutionContext.CallbackWrapper(Object obj)
   at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state)
--- End of stack trace from previous location ---
   at MS.Internal.CulturePreservingExecutionContext.Run(CulturePreservingExecutionContext executionContext, ContextCallback callback, Object state)
   at System.Windows.Threading.DispatcherOperation.Invoke()
   at System.Windows.Threading.Dispatcher.ProcessQueue()
   at System.Windows.Threading.Dispatcher.WndProcHook(IntPtr hwnd, Int32 msg, IntPtr wParam, IntPtr lParam, Boolean& handled)
   at MS.Win32.HwndWrapper.WndProc(IntPtr hwnd, Int32 msg, IntPtr wParam, IntPtr lParam, Boolean& handled)
   at MS.Win32.HwndSubclass.DispatcherCallbackOperation(Object o)
   at System.Windows.Threading.ExceptionWrapper.InternalRealCall(Delegate callback, Object args, Int32 numArgs)
   at System.Windows.Threading.ExceptionWrapper.TryCatchWhen(Object source, Delegate callback, Object args, Int32 numArgs, Delegate catchHandler)
   at System.Windows.Threading.Dispatcher.LegacyInvokeImpl(DispatcherPriority priority, TimeSpan timeout, Delegate method, Object args, Int32 numArgs)
   at MS.Win32.HwndSubclass.SubclassWndProc(IntPtr hwnd, Int32 msg, IntPtr wParam, IntPtr lParam)
   at MS.Win32.UnsafeNativeMethods.DispatchMessage(MSG& msg)
   at System.Windows.Threading.Dispatcher.PushFrameImpl(DispatcherFrame frame)
   at System.Windows.Threading.Dispatcher.PushFrame(DispatcherFrame frame)
   at System.Windows.Threading.Dispatcher.Run()
   at System.Windows.Application.RunDispatcher(Object ignore)
   at System.Windows.Application.RunInternal(Window window)
   at System.Windows.Application.Run(Window window)
   at Eto.Wpf.Forms.ApplicationHandler.Run()
   at Eto.Forms.Application.Run(Form mainForm)
   at OpenTabletDriver.UX.App.Run(String platform, String[] args)
   at OpenTabletDriver.UX.Wpf.Program.Main(String[] args)

and the application error

Faulting application name: OpenTabletDriver.Daemon.exe, version: 0.5.3.3, time stamp: 0x609308e7
Faulting module name: ntdll.dll, version: 10.0.19041.1288, time stamp: 0xa280d1d6
Exception code: 0xc0000374
Fault offset: 0x00000000000ff199
Faulting process id: 0x387c
Faulting application start time: 0x01d7e3c385981794
Faulting application path: G:\OpenTabletDriver.win-x64\OpenTabletDriver.Daemon.exe
Faulting module path: C:\Windows\SYSTEM32\ntdll.dll
Report Id: 3397db8f-5b51-4d0a-a9c1-5a45c666db3c
Faulting package full name: 
Faulting package-relative application ID:

Mrcubix, 2021-11-27:
pretty sure the second thing is different from the first one, so as a last bet, run the daemon in cmd enable touch and get the stacktrace from the console

InfinityGhost, 2021-11-29:
The wrong executable stack trace was provided, you grabbed it for `OpenTabletDriver.UX.Wpf.exe` rather than `OpenTabletDriver.Daemon.exe` @Doksha

gonX, 2021-12-07:
Closing issue as abandoned. Let us know if you still need help and we can reopen it.

Doksha, 2022-04-20:
My tablet has been broken for a while, but got fixed and the issue is still there, I tried it and I think I know what to grab from event viewer. (I get a couple different errors sometimes so I will paste them here)

Error 1 (app crashes when I hit apply after making a change)

Faulting application name: OpenTabletDriver.Daemon.exe, version: 0.6.0.3, time stamp: 0x61954b73
Faulting module name: coreclr.dll, version: 6.0.322.12309, time stamp: 0x6216cac2
Exception code: 0xc0000005
Fault offset: 0x00000000000a7faa
Faulting process id: 0x290c
Faulting application start time: 0x01d85506e0484e76
Faulting application path: G:\OpenTabletDriver.win-x64\OpenTabletDriver.Daemon.exe
Faulting module path: C:\Program Files\dotnet\shared\Microsoft.NETCore.App\6.0.3\coreclr.dll
Report Id: 4716dd42-4ed8-44ea-a67e-9640435b84b3
Faulting package full name: 
Faulting package-relative application ID: 

Error 2 (app crashes after waiting a while)

Faulting application name: bad_module_info, version: 0.0.0.0, time stamp: 0x00000000
Faulting module name: unknown, version: 0.0.0.0, time stamp: 0x00000000
Exception code: 0xc0000005
Fault offset: 0x00007ffcdd527515
Faulting process id: 0x55d4
Faulting application start time: 0x01d8550745e1a19f
Faulting application path: bad_module_info
Faulting module path: unknown
Report Id: 9e2a3648-1fc3-4711-924f-170c5f6a7d05
Faulting package full name: 
Faulting package-relative application ID: 

Sometimes if im lucky it will fist lose the tablet and will wait a few seconds to fully crash

jamesbt365, 2022-04-20:
otd logs not dotnet logs please, should be a way of seeing the exact log instead of all the information about it.

Doksha, 2022-04-21:
This is an issue that has happened to some extent on every device I have tried it on, maybe someone who actually knows what logs to grab could try it because me trying to figure out what is needed doesnt seem like its going to get anywhere

if I am lucky and OTD doesnt full crash I get this error, although its likely not at all helpful

[NullReferenceException:Error]	Object reference not set to an instance of an object.
   at OpenTabletDriver.Plugin.Output.AbsoluteOutputMode.OnOutput(IDeviceReport report)
   at OpenTabletDriver.Devices.DeviceReader`1.Main()

jamesbt365, 2022-04-21:
by the way my last message was i didn't want the crash dump i wanted the actual otd error message. Though it seems like something is probably dying because of something that the parser didn't expect. What tablet do you have? A15 isn't very descriptive, veikk?

Doksha, 2022-04-21:
yes, veikk A15
I cant really get an error message since it crashes, and apparently the log files it creates arent even useful, there isnt much I can do.

gonX, 2022-11-07:
Is this issue still present on the latest 0.6.0.4 release?

X9VoiD, 2022-12-31:
This should be fixed now with latest Touch Emu, but regarless, this is not an OTD issue so closing...

closed 2022-12-31 by X9VoiD