OpenTabletDriver/OpenTabletDriver #1673: OpenTabletDriver didn't show in programs list in windows

zooxs, 2021-11-28:
## Description

Whenever i'm trying to open OpenTabletDriver i'v to installed it instead searching it in windows search features and run it.
## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System         | Windows 10
| OpenTabletDriver Version | 0.4.1
| Tablet                   | Huion H430P

jamesbt365, 2021-11-28:
1, your opentabletdriver version is outdated and 2 it isn't in the windows run as opentabletdriver is never installed.

Mrcubix, 2021-11-28:
0.4.1 and you're trying to use the installer which is deprecated and unsupported, use the portable version instead https://github.com/OpenTabletDriver/OpenTabletDriver/releases/download/v0.5.3.3/OpenTabletDriver.win-x64.zip

closed 2021-11-28 by Mrcubix

zooxs, 2021-11-28:
Thanks @Mrcubix it's worked now. And thanks too @jamesbt365