OpenTabletDriver/OpenTabletDriver #1674: OTD Console doesn't know the "help" command

jamesbt365, 2021-11-28:
## Description

## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System         | Arch Linux
| OpenTabletDriver Version |  181b77b
| Tablet                   | XP-Pen Star G640

Doesn't know the help command yet shows help anyway.

[image]

InfinityGhost, 2021-11-29:
`help` isn't a command, rather its an argument. We could add the command, however it is a little redundant.

gonX, 2022-01-02:
There seems to be consensus that this isn't worth changing. Closing as `wontfix`.

[Discord convo](https://discord.com/channels/615607687467761684/628651971267788800/927257389982433312)

closed 2022-01-02 by gonX