OpenTabletDriver/OpenTabletDriver #1680: libinput has its own pressure offset that we should work around

gonX, 2021-11-30:
## Description

When in Linux Artist Mode, I need to apply more than 5% of my threshold (post-threshold slider) before libinput registers it as a tip down event.

Since OTD is powerful enough to have a threshold slider and remap the values (so that 51% pressure on a 50% threshold reads as 2%), I think that's a good enough reason for us to want to circumvent this tip detection.

The reason why this pressure offset exists in libinput is explained here: https://wayland.freedesktop.org/libinput/doc/latest/tablet-support.html#pressure-offset-on-worn-out-tools

A workaround is to add the `AttrPressureRange` libinput quirk for the tablet, but that means it can only be fixed in packaging.

[libinput device quirks](https://wayland.freedesktop.org/libinput/doc/latest/device-quirks.html) states:
> AttrPressureRange=N:M
>
>    Specifies the touch pressure required to trigger a press (N) and to trigger a release (M)

This might effectively not be an issue if we're interested in keeping the libinput behavior.

## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System         | Linux (Arch)
| OpenTabletDriver Version | 53a7dcf
| Tablet                   | Assumed all (but GD-0608)

X9VoiD, 2022-12-31:
Would `AttrPressureRange=1:1` mean instant press and release? We should control the "tip press" stuff from OTD to provide more flexibility.

gonX, 2025-12-05:
Yeah I don't think this is something that should be fixed on our end anyway. I can confirm that pen pressure seems a lot less mangled on newer versions of libinput, so I'm going to close this as WONTFIX/NOTOURBUG

I also seem to recall that libinput generally discourages packaging/application use of libinput quirks. A separate issue should be made that tracks removing the quirks file from packaging where relevant again. Basically if the distro uses a version of libinput earlier than 1.29 we should still include the current quirks file disabling their smoothing.

closed 2026-02-07 by Kuuuube