OpenTabletDriver/OpenTabletDriver #1686: Support Request:

ianabney, 2021-12-04:
# Support Request
I'm running Linux PopOS 21.04 and the Wacom module is being loaded every time I boot up and I have to run rmmod wacom in order to unload it. Once I do this I don't have any problems with ghost flickering of the pointer.

## What I've already tried
I've followed the guide in the linux section and have added "blacklist wacom" to the /etc/modprobe.d/blacklist.conf file. Even with this something is loading the module and I'm not sure how to find out what is loading it.

## Diagnostic Information
There are a bunch of files in the /etc/modprobe.d/ dir but most of them just blacklist different modules. I've attached a file from the directory that I think is what is loading the wacom module but I don't understand exactly what it's doing.
[alsa-base.conf.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7653480/alsa-base.conf.txt)

Please let me know if this is the correct place to post this question and I can provide other files and diagnostics if needed.

jamesbt365, 2021-12-04:
This isn't an otd issue but you could use something to autorum the command on startup

closed 2021-12-04 by jamesbt365

ianabney, 2021-12-04:
The solution was to run "update-initramfs -u" This makes the blacklist take effect properly