OpenTabletDriver/OpenTabletDriver #1687: Monitor area "Lock to usable area" setting does not work

Kuuuube, 2021-12-05:
## Description

Enabling `Lock to usable area` on monitor area does nothing. The monitor area can still be dragged out of bounds as if it were not enabled.

## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System         | Win10
| OpenTabletDriver Version | 99a1a4893a64a4191f3b0d2abf1c242f831d836b

InfinityGhost, 2021-12-05:
How did this even happen? There's no reason for this to not work lol

Kuuuube, 2021-12-05:
No wacky reproduction steps here... Literally just load up OTD and mess with the monitor area.

InfinityGhost, 2021-12-05:
Regression caused by 4ce1cf205a39188d4e5f82ca2bd3bfa39be48531

closed 2021-12-05 by InfinityGhost

aDelightz, 2022-04-28:
are you sure this is working in v0.6.0.3? it doesnt seem to for me. if you check and it is working as normal then im probably just misunderstanding the feature and you can ignore this

PR #1691 (2021-12-05): Hook display area constraint handler
- Fixes #1687.