OpenTabletDriver/OpenTabletDriver #1689: Add configuration linter

InfinityGhost, 2021-12-05:
## Description

A configuration linter would improve the configuration workflow to avoid cases like [this](https://github.com/OpenTabletDriver/OpenTabletDriver/pull/1679#pullrequestreview-822969719) so that all configuration changes would have to clean in order to be merged.

Suggestions are welcome

gonX, 2022-09-04:
Probably my bad for not searching, this seems like you want to check values for correctness? E.g. Xp-Pen -> XP-Pen and Veikk -> VEIKK?

I made another issue in #2421 that just wants to lint and make sure the formatting is good on json files, so I don't think they're exactly the same.

X9VoiD, 2022-09-04:
> This seems like you want to check values for correctness? E.g. Xp-Pen -> XP-Pen and Veikk -> VEIKK?

This would be hard to do and maintain once it's up. We already do have some checks tho: if the config can be parsed by OTD, if it refers to an existent parser, and if it's not conflicting with other configs.

Closing in favor of #2421, re-open again if this will be repurposed as an issue to add value-based linting.

gonX, 2022-11-05:
I believe there was interest in having value-based linting now

Such as having MaxX/MaxY always be floating point, while `jq` will display as integers if the float is 0.

 @InfinityGhost  confirm?

gonX, 2023-11-25:
With #3027 we are now checking config values, though not strictly int-vs-float. I plan to fix this the coming days so that the entire file is linted and matches output produced by the JSON library we use in dotnet (so that `dotnet test` grabs this).

closed 2023-11-27 by X9VoiD