OpenTabletDriver/OpenTabletDriver #1702: Support Request: Build failing

TheAtomicBen, 2021-12-06:
# Support Request
After cloning the repo and then moving into the folder to build the driver, the build fails with the error: Could not find a part of the path '.../OpenTabletDriver/OpenTabletDriver/Configurations'

My Gaomon PD1161 is still not detected and the cursor is disconnected from the pen

Is there a fix for this?

Here is a screenshot of the traceback
[image]

## What I've already tried

I haven't found any references on the build fail, but I did try moving a Configuration file into the folder from the repo, which did not work.

I also tried to see if there was a Wacom driver interfering with anything, which there wasn't.

I'm not really sure what to do next.

OS: Ubuntu 21.10
Tablet: Gaomon PD1161

## Diagnostic Information

[OpenTabletDriverDiagnostics.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7662993/OpenTabletDriverDiagnostics.txt)

InfinityGhost, 2021-12-06:
What build instructions are you following? This looks like you're trying to build the udev rules.

Mrcubix, 2021-12-06:
the udev project is deprecated as far as i'm aware, tho you should be able to get the latest udevs as of now [here](https://github.com/OpenTabletDriver/OpenTabletDriver/suites/4562453983/artifacts/122607911)

InfinityGhost, 2021-12-06:
> the udev project is deprecated as far as i'm aware, tho you should be able to get the latest udevs as of now [here](https://github.com/OpenTabletDriver/OpenTabletDriver/suites/4562453983/artifacts/122607911)

The udev project itself isn't, the old repository is. It should instead be run through `./generate-rules.sh` when you clone the repository.

TheAtomicBen, 2021-12-06:
I was following the instructions at https://osu.ppy.sh/community/forums/topics/1340468?n=1

Mrcubix, 2021-12-06:
[image]
first thing i see

TheAtomicBen, 2021-12-06:
OMG I FEEL LIKE AN IDIOT

Mrcubix, 2021-12-06:
Also for your tablet, you should be able to make use of the plugin called `OTD.Backport` in stable (package `opentablerdriver`), either get it in the Plugin manager or get a more recent version [here](https://github.com/Mrcubix/OTD.Backport/releases/tag/1.0.2) and drag and drop the downloaded zip file as is in the Plugin manager

closed 2021-12-06 by InfinityGhost