OpenTabletDriver/OpenTabletDriver #1703: NRE when migrating from 0.5.x to 0.6.0

X9VoiD, 2021-12-06:
## Description

Logs say migration is successful but will result to an NRE crash later during output mode configuration.

InfinityGhost, 2021-12-06:
Trace?

andigandi, 2021-12-10:
You mean this? Happens when I plug in my tablet on current master branch, using dotnet 6.0

```
Unhandled exception. System.NullReferenceException: Object reference not set to an instance of an object.
   at OpenTabletDriver.Desktop.Profiles.AreaSettings.GetDefaults(DigitizerSpecifications digitizer)
   at OpenTabletDriver.Desktop.Profiles.AbsoluteModeSettings.GetDefaults(DigitizerSpecifications digitizer)
   at OpenTabletDriver.Desktop.Profiles.Profile.GetDefaults(TabletReference tablet)
   at OpenTabletDriver.Desktop.Profiles.ProfileCollection.Generate(TabletReference tablet)
   at OpenTabletDriver.Desktop.Profiles.ProfileCollection.GetProfile(TabletReference tablet)
   at OpenTabletDriver.Daemon.DriverDaemon.SetSettings(Settings settings)
   at OpenTabletDriver.Daemon.DriverDaemon.<.ctor>b__0_2(Object sender, DevicesChangedEventArgs args)
   at System.Threading.Tasks.Task.<>c.<ThrowAsync>b__127_1(Object state)
   at System.Threading.QueueUserWorkItemCallbackDefaultContext.Execute()
   at System.Threading.ThreadPoolWorkQueue.Dispatch()
   at System.Threading.PortableThreadPool.WorkerThread.WorkerThreadStart()
   at System.Threading.Thread.StartCallback()
```

InfinityGhost, 2021-12-23:
Unable to reproduce as of ed3cb17a3bfb0f960ccc9d6f18f273073b96d352

closed 2021-12-23 by InfinityGhost