OpenTabletDriver/OpenTabletDriver #1707: GAOMON M10K PRO

zakaria-chahboun, 2021-12-07:
Hey! I don't find my TABLET!
And yeah i use the final release!
it's `GAOMON M10K PRO`
i already reported about it #590 

[image]

jamesbt365, 2021-12-07:
ALL CAPS doesn't help YOUR TABLET may be SUPPORTED ON PRE RELEALSE, check the BACKPORT PLUGIN in the PLUGIN MANAGER

Mrcubix, 2021-12-07:
# Allow for detection of tablets that require backporting

- Download OTD.Backport.zip from [here](https://github.com/Mrcubix/OTD.Backport/releases/latest)
- Open OTD release (v0.5.3.3)
- Go to Plugins > Open Plugin Manager
- Drag and drop OTD.Backport.zip into the Plugin Manager window
- Go to Tools > OTD.Backport
- Check the Enable box
- Click Save
- Do Tablets > Detect tablet

closed 2021-12-07 by Mrcubix