OpenTabletDriver/OpenTabletDriver #1709: Add support for XP-Pen Deco Pro Medium

andigandi, 2021-12-08:
# Tablet Information

| Name            | Value |
| --------------- | ----- |
| Manufacturer    | XP-Pen
| Product Name    | Deco Pro Medium / Deco Pro MW
| Width           | 11"
| Height          | 6"
| Pressure Levels | 8192 levels
| LPI             | 5080 LPI

## Diagnostic Information

The device has USB and Bluetooth, `lsusb` shows these things for USB connection: `ID 28bd:0934 XP-Pen Deco Pro MW`. I have tried to dump all strings, but it keeps saying "Please replug the tablet"... I also tried to manually add the JSON (based on the non-wireless, changed USB IDs) but `otd` does not detect it. Is there any further information, that you need?

[diagnostics.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7676103/diagnostics.txt)
[Deco Pro Medium Wireless.json](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7676106/Deco.Pro.Medium.Wireless.txt)

Mrcubix, 2021-12-08:
keep replugging and pressing ok, it will eventually reach past the string that reset the tablet into bootloader mode, in the meanwhile i'll be looking at what i can do

andigandi, 2021-12-08:
Thank you for the quick reply!

So just that I get this right, I really unplug the USB, plug it in again and click on OK, again and again so eventually it will work? Is this a matter of timing, or count or something? When I plug in, first the tablet's corners light, then go off and then the red ring glows. Does it matter when I press OK during this process?

Mrcubix, 2021-12-08:
you just need to do that as much as it ask

Mrcubix, 2021-12-08:
Shouldn't matter, press OK

andigandi, 2021-12-08:
Okay, it's because I did this process about 20 times and now I am wondering, if I'm doing something wrong.

Mrcubix, 2021-12-08:
it could be possible that the rest of the string pas a certain point reboot into bootloader but that would be weird
anyway, use install the package `opentabletdriver` instead and try this config, place it in `/usr/share/OpenTabletDriver/Configurations`
[Deco Pro Medium Wireless.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7676849/Deco.Pro.Medium.Wireless.zip)

andigandi, 2021-12-08:
Thanks, but no luck with that configuration. Because Arch switched to Mono 6, I cannot create the `opentabletdriver` package from AUR but have to use `opentabletdriver-git` instead. Might this cause problems?

Just to make sure, is this the correct udev rule? I created it in /etc/udev/rules.d/99-xp-pen.rules and run `udevadm control --reload-rules`

```
# XP-Pen Deco Pro Medium
SUBSYSTEM=="hidraw", ATTRS{idVendor}=="28bd", ATTRS{idProduct}=="0934", MODE="0666"
SUBSYSTEM=="usb", ATTRS{idVendor}=="28bd", ATTRS{idProduct}=="0934", MODE="0666"
SUBSYSTEM=="input", ATTRS{idVendor}=="28bd", ATTRS{idProduct}=="0934", ENV{LIBINPUT_IGNORE_DEVICE}="1"
```

Sublimelime, 2021-12-08:
> Because Arch switched to Mono 6, I cannot create the opentabletdriver package from AUR but have to use opentabletdriver-git instead.

You can also just grab the precompiled tarball ("portable installation") from the releases here on github if you can't build the stable AUR package.

jamesbt365, 2021-12-08:
You mean dotnet?

andigandi, 2021-12-08:
Yes, I meant dotnet. Sorry.

```
% dotnet --version
5.0.403
```

Mrcubix, 2021-12-08:
I don't know about arch bs, but you could probably make use of the linux tar available for 0.5.3.3, if not you'll have to build it yourself

jamesbt365, 2021-12-08:
Download these from the aur then try again dotnet-runtime-5.0-bin dotnet-sdk-bin

andigandi, 2021-12-08:
That's what I did. Installed dotnet 5 and used the precompiled tarball. I untarred it to my home folder and ran it from console. I am getting the output `[Detect:Info]   Searching for tablet 'XP-Pen Deco Pro Medium Wireless'`, that means the config file is in the right place?

dotnet-sdk-5.0-bin 5.0.12.sdk403-1
dotnet-runtime-5.0-bin 5.0.12.sdk403-1

Mrcubix, 2021-12-08:
Export a diag, something might have changed

andigandi, 2021-12-08:
[diag.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7677811/diag.txt)

Mrcubix, 2021-12-08:
use this config https://github.com/OpenTabletDriver/OpenTabletDriver/files/7677945/Deco_Pro_Medium_Wireless.zip

andigandi, 2021-12-08:
Thank you! It is working with this one! I am currently playing around. Pen, Pen Buttons, sensitivity and the 8 auxilliary buttons seem to be working. 

Is it possible to use the device via bluetooth with OpenTabletDriver or is only USB supported?

Mrcubix, 2021-12-08:
export a diag with bluetooth, i thought this was with bluetooth enabled since the Deco pro medium is already supported normally

andigandi, 2021-12-09:
This is diag with bluetooth connection. But I am not sure if this is usable. From what I can see, the PID and VID for the bluetooth connected device seem to be the same. Same udev rule as shown above, but still libinput grabs the tablet. Also, I can not read device strings. Permission problem? But that should have been handled by the udev rule?

[diag-bt.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7683637/diag-bt.txt)

gonX, 2022-11-07:
I don't see the Bluetooth device in the last provided diagnostic. Can you try uploading a new one from the latest release (0.6.0.4)?

cstrahan, 2023-07-14:
If it helps the devs, you could look at how this tablet is handled in [`userspace-tablet-driver-daemon`](https://github.com/kurikaesu/userspace-tablet-driver-daemon): https://github.com/kurikaesu/userspace-tablet-driver-daemon/pull/56

As an owner of this tablet, I would love to have support for it in OTD :)

cstrahan, 2023-07-15:
I tried making a modified copy of the existing "XP-Pen Deco Pro Medium" config and dropping it in `~/.local/share/OpenTabletDriver/Configurations` (with the `ProductID` replaced appropriately):

```json
{
  "Name": "XP-Pen Deco Pro Medium Wireless",
  "Specifications": {
    "Digitizer": {
      "Width": 278.99,
      "Height": 156.995,
      "MaxX": 55798.0,
      "MaxY": 31399.0
    },
    "Pen": {
      "MaxPressure": 8191,
      "Buttons": {
        "ButtonCount": 2
      }
    },
    "AuxiliaryButtons": {
      "ButtonCount": 8
    },
    "MouseButtons": null,
    "Touch": null
  },
  "DigitizerIdentifiers": [
    {
      "VendorID": 10429,
      "ProductID": 2356,
      "InputReportLength": 10,
      "OutputReportLength": 10,
      "ReportParser": "OpenTabletDriver.Configurations.Parsers.XP_Pen.XP_PenReportParser",
      "FeatureInitReport": null,
      "OutputInitReport": [
        "ArAE"
      ],
      "DeviceStrings": {},
      "InitializationStrings": []
    }
  ],
  "AuxilaryDeviceIdentifiers": [],
  "Attributes": {
    "libinputoverride": "1"
  }
}
```

... but the tablet is still not detected.

It's clearly *trying* to detect it:

```
$ journalctl --user -u opentabletdriver.service | grep Wireless
Jul 14 18:48:51 pop-os otd-daemon[16996]: [Detect:Debug]        Searching for tablet 'XP-Pen Deco Pro Medium Wireless'
Jul 14 18:53:58 pop-os otd-daemon[16996]: [Detect:Debug]        Searching for tablet 'XP-Pen Deco Pro Medium Wireless'
Jul 14 18:54:00 pop-os otd-daemon[16996]: [Detect:Debug]        Searching for tablet 'XP-Pen Deco Pro Medium Wireless'
Jul 14 19:01:43 pop-os otd-daemon[20174]: [Detect:Debug]        Searching for tablet 'XP-Pen Deco Pro Medium Wireless'
```

Any ideas? Or anything I can provide from my end?

cstrahan, 2023-07-15:
I think I have it working no! (at least wired, so far)

I took the `InputReportLength`/`OutputReportLength` values from here: https://github.com/OpenTabletDriver/OpenTabletDriver/issues/1709#issuecomment-989122253

And now I have:

```json
{
  "Name": "XP-Pen Deco Pro Medium Wireless",
  "Specifications": {
    "Digitizer": {
      "Width": 278.99,
      "Height": 156.995,
      "MaxX": 55798.0,
      "MaxY": 31399.0
    },
    "Pen": {
      "MaxPressure": 8191,
      "Buttons": {
        "ButtonCount": 2
      }
    },
    "AuxiliaryButtons": {
      "ButtonCount": 8
    },
    "MouseButtons": null,
    "Touch": null
  },
  "DigitizerIdentifiers": [
    {
      "VendorID": 10429,
      "ProductID": 2356,
      "InputReportLength": 10,
      "OutputReportLength": 10,
      "ReportParser": "OpenTabletDriver.Configurations.Parsers.XP_Pen.XP_PenReportParser",
      "FeatureInitReport": null,
      "OutputInitReport": [
        "ArAE"
      ],
      "DeviceStrings": {},
      "InitializationStrings": []
    }
  ],
  "AuxilaryDeviceIdentifiers": [],
  "Attributes": {
    "libinputoverride": "1"
  }
}
```

[image]

jamesbt365, 2023-07-15:
> (at least wired, so far)

Bluetooth is not supported because a limitation in a dependency upstream.

cstrahan, 2023-07-15:
> Bluetooth is not supported because a limitation in a dependency upstream.

@jamesbt365 which "upstream" are you referring to here?

cstrahan, 2023-07-15:
Things *mostly* work wired now, but I run into the issue I had in `userspace-tablet-driver-daemon` where pressing one of the eight buttons causes that button to be spammed until you unplug the tablet, as described here: https://github.com/kurikaesu/userspace-tablet-driver-daemon/pull/56

This was the fix: https://github.com/kurikaesu/userspace-tablet-driver-daemon/commit/64a1f042bd6b012a321c4a5c8b4b0d02f49d3fb6

So until that's resolved, I can't use any of the 8 buttons on the tablet :disappointed: 

The XP-Pen Deco mini7 is cheap enough (and in retrospect, the size is actually probably better for my purposes), so I'm going to try giving that a go instead. Basically want a tablet for doodling screen for presentation purposes, or hashing things out during interviews, etc.

cstrahan, 2023-07-15:
To port the aforementioned fix:

```c++
        // Not exactly what this message is but the deco pro wireless seems to send it randomly so let's ignore it for now
        if (data[1] == 0xf2 && data[3] == 0x63) {
            return;
        }
```

Should something like this suffice (in [`OpenTabletDriver.Configurations/Parsers/XP_Pen/XP_PenReportParser.cs`](https://github.com/OpenTabletDriver/OpenTabletDriver/blob/63de489815288d8ccac434d59ef41b6bba20379f/OpenTabletDriver.Configurations/Parsers/XP_Pen/XP_PenReportParser.cs))?

```c#
using OpenTabletDriver.Tablet;

namespace OpenTabletDriver.Configurations.Parsers.XP_Pen
{
    public class XP_PenReportParser : IReportParser<IDeviceReport>
    {
        public IDeviceReport Parse(byte[] report)
        {
            if (report[1] == 0xC0)
                return new OutOfRangeReport(report);

            if (report[1] == 0xF2 && report[3] == 0x63)  // <=== new code
                return new OutOfRangeReport(report) ;    // <=== new code

            if (report[1].IsBitSet(4))
                return new XP_PenAuxReport(report);

            if (report.Length >= 12)
                return new XP_PenTabletOverflowReport(report);
            else if (report.Length >= 10)
                return new XP_PenTabletReport(report);
            else
                return new TabletReport(report);
        }
    }
}
```

cstrahan, 2024-02-02:
For posterity, on macOS I had to add the configuration file to `/Applications/OpenTabletDriver.app/Contents/MacOS/Configurations`. I don't see this documented anywhere (and I wasn't able to track that down from grepping through the source within ~5 minutes or so) -- had to infer that from checking the process list (on a hunch) and finding that the daemon was invoked like so:

`/Applications/OpenTabletDriver.app/Contents/MacOS/OpenTabletDriver.Daemon -c /Applications/OpenTabletDriver.app/Contents/MacOS/Configurations`

rkolbaskin, 2025-05-28:
@cstrahan report[1] == 0xF2 seems to be a battery level report with report[3] byte being battery level, so you shouldn't check report[3] value.