OpenTabletDriver/OpenTabletDriver #1710: Button press is read as press+relase in AHK

uszymyszy, 2021-12-08:
## Description
When using Autohotkey to write macros, the original Wacom driver sends "Press" event when button is pressed and "Release" when released. OpenTablet sends "Press+Release" when pressed and "Release" when released. In other programs all presses works OK.

## System Information:

| Win 10  | Wacom Bamboo One    |

gonX, 2021-12-08:
Which version of OpenTabletDriver is this?

jamesbt365, 2021-12-08:
This isn't really an otd issue? Just change your ahk script. Unless somebody else has a solution

Mrcubix, 2021-12-08:
Afaik, OTD make use of what your tablet send, so if it send false for a specific express key right after sending true then it's the device fault, not the driver.

you would on top of that need to check if it flicker as that would indicate that wacom make use of some sort of debounce on their end.

jamesbt365, 2021-12-08:
replace the config with this one:
[Deco_Pro_Medium_Wireless.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/7677945/Deco_Pro_Medium_Wireless.zip)

uszymyszy, 2021-12-08:
Yes, I changed the button reading in AHK to temporary fix this issue. AHK has several modes to read the buttons releases and the "Keywait" is default and faster and "key up" is an alternative. In Wacom driver both modes works as expected while in OTD only second one works. The Deco Pro Medium zip does not works and hangs the driver. I'm using OTD v.0.5.3.3 on Wacom CTL-671 Medium, Win 10 x64

gonX, 2022-11-07:
As mentioned, OpenTabletDriver will forward aux buttons as-is from the device. Please open a new issue if this is still happening, since if the original driver has the functionality, it hints to that we're parsing aux buttons incorrectly for that specific tablet. However, if the tablet does not report a keyup (separate or not) there is no way for us to fix this on those tablets.

closed 2022-11-07 by gonX