OpenTabletDriver/OpenTabletDriver #1712: Support Request: GAOMON S620

csc840219, 2021-12-10:
# Support Request

 I've follow the steps from the wiki to install the driver on Linux (I'm using elementary os 6.0, kernel 5.11).
But tablet does not work.
## What I've already tried

I tried the "Failed to open device streams" section from FAQ.
At least I tried this from https://osu.ppy.sh/community/forums/topics/1340468?n=1
    git clone https://github.com/InfinityGhost/OpenTabletDriver-udev.git
    cd OpenTabletDriver-udev
    git submodule update --init --remote
./build.sh
    sudo mv ./build/99-opentabletdriver.rules /etc/udev/rules.d/99-opentabletdriver.rules
    sudo udevadm control --reload-rules
After reboot the tablet works, but the app stills saying "Not detected tablet" and "not permissions for open a stream". So I can not configure or test the tablet on the app but it is working.
Any idea?
## Diagnostic Information

jamesbt365, 2021-12-10:
That is outdated

use this
http://opentabletdriver.net/Wiki/FAQ/Linux

csc840219, 2021-12-10:
I've already tried it, but nothing from there works for me.

http://opentabletdriver.net/Wiki/FAQ/Linux

jamesbt365, 2021-12-10:
`git clone https://github.com/OpenTabletDriver/OpenTabletDriver.git
cd ./OpenTabletDriver
sudo mv ./bin/99-opentabletdriver.rules /etc/udev/rules.d/99-opentabletdriver.rules
sudo udevadm control --reload-rules
cd ..
rm -rf OpenTabletDriver`

literally this script

csc840219, 2021-12-10:
I've already tried that, but by the time I replace "/etc/udev/rules.d/99-opentabletdriver.rules" and reboot, the tablet stops working.

jamesbt365, 2021-12-10:
try it again, i made a typo and have edited it.

jamesbt365, 2021-12-10:
then after help > export diag in otd and send that file here

Mrcubix, 2021-12-10:
it's all cool and all but i don't see the most fundamental piece of information that's also part of the template: the diagnostic file

csc840219, 2021-12-12:
Hello. 
At last I tried something different. I've read that this tablet is detected by default and there is no need for any other software.
I tried "https://github.com/bigbigmdm/GAOMON-S620-in-LINUX" and it works ok. Also I could config all buttons.
Thanks for your time and attention.

closed 2021-12-12 by csc840219