OpenTabletDriver/OpenTabletDriver #1713: GAOMON S620 not detected on app.

csc840219, 2021-12-10:
## Description

I've follow the steps from the wiki to install the driver on Linux (I'm using elementary os 6.0, kernel 5.11).
But tablet does not work.
I tried the "Failed to open device streams" section from FAQ.
At least I tried this from https://osu.ppy.sh/community/forums/topics/1340468?n=1
git clone https://github.com/InfinityGhost/OpenTabletDriver-udev.git
cd OpenTabletDriver-udev
git submodule update --init --remote
./build.sh
sudo mv ./build/99-opentabletdriver.rules /etc/udev/rules.d/99-opentabletdriver.rules
sudo udevadm control --reload-rules
After reboot the tablet works, but the app stills saying "Not detected tablet" and "not permissions for open a stream". So I can not configure or test the tablet on the app but it is working.
Any idea?
## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System         |
Elementary Os 6.0 (kernel 5.11)
| OpenTabletDriver Version |
| Tablet                   |
Gaomon S620.

jamesbt365, 2021-12-10:
duplicate of #1712

please refrain from making several issues for the same problem

jamesbt365, 2021-12-10:
Also this is a **outdated** script as i mentioned.

closed 2021-12-10 by jamesbt365