OpenTabletDriver/OpenTabletDriver #1714: Include installation source, build date and packaging date in diagnostics

gonX, 2021-12-10:
Adding this would allow helpers to track down or quickly identify issues with packaging, for example the notorious udev solution that's being posted on Discord regularly is likely caused by a package being broken, or people still using the broken 0.5.3.3 .deb file. It is also impossible to know whether a user is using the .deb file, or the .tar.gz file, the latter of which does not include udev rules.

I propose for Linux packagers to implement adding a consistently named file into the OpenTabletDriver package which should include:
- Maintainer name (if relevant, e.g. github for actions build)
- Distro/packaging system (arch/deb/yum/generic/etc)
- Packaging date (I propose ISO8601 but Unix epoch can work too).

The name can be something along the lines of `packager-info.json` but I'm open for ideas

Example content:
```json
{
  "MaintainerName": "gonX",
  "PackagingSystem": "AUR opentabletdriver-git",
  "PackageDate": 1639137233
}
```
Generated by a oneliner: `echo '{"MaintainerName": "gonX", "PackagingSystem": "AUR opentabletdriver-git", "PackagingDate": '$(date +%s)'}'`

Actual file format and layout is up to the implementer - it doesn't have to be json.

For Windows we can likely forego the file and just rely on build date alone since the significant majority of people use the precompiled builds.

Things that need to be solved before this issue can be closed:
- [ ] Code in OTD to parse the file and expose it in diagnostics
- [ ] Have AUR maintainer update packages
- [ ] Have Gentoo maintainer update package(s)
- [ ] Update [build.sh](https://github.com/OpenTabletDriver/OpenTabletDriver/blob/master/build.sh)
- [ ] Update all [OpenTabletDriver.Packaging](https://github.com/OpenTabletDriver/OpenTabletDriver.Packaging) scripts
- [ ] Ensure build date is a property in OpenTabletDriver (#1672)

(let me know if I missed anything)

Bonus points if we could show these properties in About OTD (F1 hotkey) as well, so curious users can see it without exporting diags

TL;DR more info in diagnostics to help track which, when, and where the install is from

Sublimelime, 2021-12-10:
I think most of this issue could be mitigated by just making more use of git tags. The AUR git package has been repeatedly updated, but without a git tag - it's just built off an arbitrary commit where things seem okay enough.

#1672 helps this by including the source revision ID (being the current commit SHA) in builds made from build.sh or from actions (@gonX  would need to correct me, but isn't the aur package using build.sh to build too? If not, it maybe should, we need to centralize things and not have any package doing special things other release mediums don't). That already fixes things for the deb as debs are precompiled, so the build date matches the packaging date.

Starting to find regular "stopping points" and tagging them as something like "0.6.0-pre1", "0.6.0-pre2", etc and then just updating the aur package each time a new tag is made (and also updating OTD's internal version to the same "pre1", "pre2", etc) would be far simpler. We haven't made a proper prerelease in a long time (since september).

> I propose for Linux packagers to implement adding a consistently named file into the OpenTabletDriver package which should include:

I don't really like the idea of just having an echo command somewhere along the deployment that OTD reads - it's a brute force solution that doesn't take advantage of dotnet's properties/csproj stuff, nor the power of git to track releases (and also breaks if someone builds OTD in some way that doesn't make that file, AND needs extra code to not run on windows/mac).

#1672 adds the build date and the current commit SHA to the properties, (which is already included in diags as something the PR also does). I could also add the current git **tag** as well, or even instead of the commit SHA - but again *the tag needs to be relevant - the most recent tag I believe is still from the stable 0.5.xxx stuff.*

> include the udev rules

I've been kinda brainstorming a sort of troubleshooter for OTD - basically if we detect a device that looks like a tablet but can't be opened because bad udev rules, either just go ahead and generate them (+install them? would need sudo tho), or tell the user how to fix automatically (pop up an error or box saying "hey we found your tablet technically but need permission to actually access it"), so they don't need to ask in the discord.

gonX, 2021-12-11:
> The AUR git package has been repeatedly updated, but without a git tag - it's just built off an arbitrary commit where things seem okay enough.

Not sure what you mean by this. AUR -git packages always pulls the latest from master. The version in AUR -git packages should be ignored, as they're updated when `makepkg` is run. I do try to update the package version when the PKGBUILD updates so it doesn't look as bad. When we eventually get a new tag it will automatically be pulling that in.
I agree we need `0.6-pre` tagged though. With the way things are tagged now it does indeed make the -git version look older at a quick glance.

> isn't the aur package using build.sh to build too?

It is. I only add versioning to the arguments, which will have to be removed once #1672 rolls around. I'm keeping up to date with it.

> I don't really like the idea of just having an echo command somewhere along the deployment that OTD reads - it's a brute force solution that doesn't take advantage of dotnet's properties/csproj [...]

That's by design, it allows repackaging without rebuilding the entire project. It would also allow us to see if packaging and build dates are mismatched, which may indicate an improper install. I'd obviously prefer if we could have both what you're asking _and_ for a way to trace Linux-specific packaging info. As I specified I don't really care about Windows, since the majority of people are using a precompiled build, so there far less need to know these extra things.

> I've been kinda brainstorming a sort of troubleshooter for OTD - basically if we detect a device that looks like a tablet but can't be opened because bad udev rules, either just go ahead and [...]

We already tell the user a direct URL how to fix their issues in some cases, but I agree more common issues should be linked to in log. `EACCESS` doesn't have handling for this, for example - but this is better covered in a separate issue.

X9VoiD, 2021-12-11:
The git tag helps with identifying the source code that the package compiled with. The extra information proposed here helps with identifying issues not necessarily within the code, but to the packaging itself.

gonX, 2021-12-20:
I'm going to go through with this if there are no objections to the proposed format. @InfinityGhost thoughts? Basically we want packaging info accessible from diagnostics, so that we know which distro and package the user is using.

jamesbt365, 2022-06-10:
Isn't some of this done now?

gonX, 2022-11-07:
I'll have to go over it to see what we're missing. I think I can achieve the rest of what'ss needed on both the OTD and Arch side, but I would have to get in touch with the other package maintainers once I have a plan set out for it.

gonX, 2023-11-15:
Can probably use `OperatingSystem.Attributes` introduced with #2944 here.